ccrisan/motionpie #54: CPU usage 100% - build 20141228

sh2515, 2015-01-02:
I noted after a 1/2 an hour that the screen refresh was taking approximately 5 seconds between stream refresh.  I ssh'd into pi and found the cpu was running at 100% and the network was running quite high - though don't know where it was streaming to. After running `top` in ssh I found
```
5697   290 root     S    83320  67%  93% /usr/bin/motion -c /data/etc/motion.conf -n -d 1
  290     1 root     S    26828  22%   2% /usr/bin/python2.7 /programs/motioneye/motioneye.py
12401   252 root     S     5564   4%   1% sshd: root@pts/1
   33     2 root     SW<      0   0%   1% [VCHIQ-0]
13596 13572 admin    S     5504   4%   0% sshd: admin@pts/0
13614 13597 admin    R      952   1%   0% top
  176     1 root     S     1888   2%   0% {S35wifi} /bin/sh /etc/init.d/S35wifi start
  353     1 root     S     1884   2%   0% {S95motioneye} /bin/sh /etc/init.d/S95motioneye start
   38     2 root     SW       0   0%   0% [kworker/u2:1]
 7140     2 root     SW       0   0%   0% [kworker/u2:0]
13824     2 root     SW       0   0%   0% [kworker/0:2]
  300   280 root     S     9524   8%   0% smbd -D
  280     1 root     S     9468   8%   0% smbd -D
13572   252 root     S     5504   4%   0% sshd: admin [priv]
  252     1 root     S     2896   2%   0% /usr/sbin/sshd
  139     1 root     S     2768   2%   0% /usr/sbin/wpa_supplicant -iwlan0 -c/data/etc/wpa_supplicant.conf -Dwext -B
   69     1 root     S     1988   2%   0% /sbin/udevd -d
12414 12401 root     S     1960   2%   0% -sh
```

sh2515, 2015-01-02:
logs
https://drive.google.com/file/d/0B-3NZeH9lwTDMGVyc1FaSml4ZDA/view?usp=sharing
dmesg
https://drive.google.com/file/d/0B-3NZeH9lwTDM2VKaDNFQlk3Qkk/view?usp=sharing

ccrisan, 2015-01-02:
Thanks for reporting the issue. Reading through the log files you provided, it seems to me that the motion daemon has to deal with a large movie file. dmesg says "motion invoked oom-killer" at one point and it's the process that eats most of the CPU. Lots of memory and CPU? probably a movie encoding.

Otherwise, if you don't have movies enabled at all, what resolution and framerate are you using?

sh2515, 2015-01-02:
I have build 20141228 motion movies on at 1280x1024. The previous stable release had it running at 1440x960 - and various others during testing - and working with approximately 50% cpu. 

stable log files
https://drive.google.com/file/d/0B-3NZeH9lwTDOVdreTZxaDRHQ00/view?usp=sharing

ccrisan, 2015-01-11:
At that resolution with area_detect, I won't expect anything below 100% on a Raspberry PI. If the CPU usage comes from motion (as it appears), there's really nothing I can do about it.

closed 2015-01-11 by ccrisan