ccrisan/motionpie #55: Cannot Send Email

sfulch1323, 2015-01-02:
Hello,
I have uploaded the latest version of MotionPie to an SD card and set it up on my RaspberryPi.
Everything is working great for me accept for one important aspect. I cannot seem to get email to send using the SMTP settings I've configured.
I've tried several things such as using different SMTP server addresses, such as Google, Hotmail, and my ISP's SMTP service with no luck. I don't think its a problem with my network or the SMTP config, as I'm using my ISP's SMTP server for backup notifications with no issues, yet when I use the exact same settings in MotionPie, it does not work.
Can you help point me in the right direction for getting this issue resolved? For example, some log file locations I can look at to see where things might be going wrong?
Thanks!
Great software by the way. You're really talented.

ccrisan, 2015-01-02:
You'll find useful logs at /var/log/motion.log and /var/log/motioneye.log, on the data partition. The former should tell us what's wrong with the email sending.

sfulch1323, 2015-01-02:
Wow, thanks for the quick response. I've got it hooked up at home, so I'll log in later and checkout those logs and post them here for your review. thanks very much.

sfulch1323, 2015-01-02:
I'm curious as well, does the auto-email attempt to attach all the pictures/video its capturing to the email? Just wondering if that could be part of the problem, if the SMTP server has an attachment limit or something.

ccrisan, 2015-01-02:
Well, it just looks for picture files with a timestamp around the motion event moment  and attaches everything that it finds within a span of around 5 seconds. Of course the number of attached pictures (if any) depends on the still image capturing configuration. I've been meaning to make these 5 seconds configurable and maybe add a limit to the number of attached pictures. It's going to happen some day :)

sfulch1323, 2015-01-03:
Good news. I got it working. There was a couple of issues. I was getting an error thrown in your python script, I don't think it liked the network share I was pointing it to to capture pictures, so I set it back to the default, which is to store it on the local storage device (default) /dev/mmcblk0b3. I wasn't getting that error anymore, but a new error specifically dealing with the SMTP server, it was "smtp auth extension not supported by server". This was with my ISP's SMTP server connection info, so I guess it didn't like that. I then changed it to googles SMTP server on port 587 with TLS and it now works. 
thanks for the help.
One last question, I noticed there's a section for working schedule where you can pick days, and when it will capture motion. I've set all of mine to ON and set every day from 00:00 to 23:30. Does this need to be set for it to email me notifications? Or will it always send email if the working schedule section is turned off? Also how come its missing a half an hour from 11:30PM to 12:00 midnight? will it always skip the half an hour?Just thought that was odd.

ccrisan, 2015-01-03:
Working schedule is not related to emails directly, but rather to motion detection in general. If the w/s however says that motion should be ignored, you won't receive emails for tbat period of time.

23:30 is just the last available option in the combo. You could press ESC and simply type in your hour and minute, e.g. 23:59. Intervals are inclusive, so it would cover the entire day.

closed 2015-01-03 by ccrisan

Voodoofly, 2015-07-27:
Hello ccrissan

at first i will say a big thx, bevor i found your image, i try so 5 hours, but only trouble

then i take your Image and it works in 5 Minutes, for this a verry verry big thank you

I wish now to replace the ip cam in my "self made"  "door plate"  with a bether resolution Raspi IP Cam 

I have buy a 5MP resolution ip cam für Raspi

in motionpie and "with the motion detection" my resolutuon options going only up to 1600x1200 
when i choose fast internet cam i can go to higher resolution but the motions detection options then missing

Now my question, are there a way in future allways have the the motion detection in all resolutuon ?

best wishes from germany

Tomislav

ccrisan, 2015-07-28:
There are no plans to add motion detection support for the Fast Network Camera.

Voodoofly, 2015-07-28:
Ok i understand  L
 
And a higher resolution  then 1600x1200 with Motion Detection ?  
 
Is the CPU Power the problem ?

ccrisan, 2015-07-29:
Yes, it's the CPU.

Petreleslie, 2015-09-23:
I too am enjoying motionpie. on RaspberryPi B+ thanks. 
Email notifications are not working from motionpie. The smtp settings DO work with another application. 
The motion log (which I can't understand) says:
Connection unexpectedly closed: timed out
2015-09-23 16:44:31: ERROR: failed to send mail: Connection unexpectedly closed: timed out
Traceback (most recent call last):
  File "/programs/motioneye/sendmail.py", line 182, in on_message
    send_mail(server, port, account, password, tls, to, subject, message, files)
  File "/programs/motioneye/sendmail.py", line 57, in send_mail
    conn = smtplib.SMTP(server, port, timeout=getattr(settings, 'SMTP_TIMEOUT', 60))
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 317, in connect
  File "/usr/lib/python2.7/smtplib.py", line 365, in getreply
SMTPServerDisconnected: Connection unexpectedly closed: timed out

Thanks for any help.

Voodoofly, 2015-09-23:
Hi,

did you try the port 587 ?

Petreleslie, 2015-09-24:
Port 587 DOES work with Motionpie in my setup. thank you.