ccrisan/motionpie #56: [Feature request] Include SSH client

NicApicella, 2015-01-03:
Hello!

Incredible work! Works like a charm.

For my needs, some more feedback-triggers when motion is detected would be really useful. I tried the "Motion Notifications", but the options are pretty limited. Since I'm aware that having a complete *nix distribution on the Pi is impractical, I thought I'd simply connect via ssh to another computer in my network and call a script—that way, the remote host could do the grunt-work instead of the Pi; whatever that "work" may be.
Unfortunately, I noticed that there doesn't even seem to be any ssh client installed! Is there any way to remedy this? (Or am I missing something altogether?)

I was thinking of setting up keys for password-less ssh-login and then running a command on the lines of `ssh user@remote ~/.bin/motiondetect-script.sh'. Would that be feasible?

Thanks and kind regards,
Nicola

ccrisan, 2015-01-03:
Yes, the ssh client can and will be included in a future build. However using webhooks might be more convenient in your case.

NicApicella, 2015-01-03:
Cool, great! Can't wait!

Yes, I am in fact toying around with webhooks; but since I'm far not as familiar with them as with ssh, I'm having a hard time. ;-)

(Oh, and should you be looking for more services to add: mDNS would be next on my wishlist! :-D )

Thanks,
Nicola

ccrisan, 2015-01-03:
Duly noted: https://github.com/ccrisan/motionPie/issues/57

closed 2015-01-11 by ccrisan