ccrisan/motionpie #58: Wifi issue using AWUS036NEH wifi adaptor

dkelley810, 2015-01-04:
Awesome job with MotionPi

my Current Version 	- 20150102

wifi dongle - ALFA , AWUS036NEH (Ralink RT3070) Tested on Debian Squeeze (with Ralink firmware package)
Working great with LAN, but unable to get it working with WIFI
my wpa_supplicant.conf -

update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
    scan_ssid=1
    ssid="2WIRE214"
    psk="**********"
}

It works great with my Raspbian SDcard in the same Pi.
Has any encountered the same issue? Any additonal information needed please contact me.

Thank You,
Dennis Kelley

dkelley810, 2015-01-04:
When plug adaptor into Pi and run lsusb the WiFi Dongle show in the list as - Bus 001 Device 005: ID 148f:3070

148f  Ralink Technology, Corp.
3070  RT2870/RT3070 Wireless Adapter - from http://www.linux-usb.org/usb.ids

ccrisan, 2015-01-05:
Could you tell us the output of the "lsmod" and "ifconfig -a" commands?

dkelley810, 2015-01-05:
[root@mp-9026ce7f ~]# lsmod
Module                  Size  Used by    Not tainted
rt2800usb              14757  0 
rt2800lib              67656  1 rt2800usb
rt2x00usb               6532  1 rt2800usb
rt2x00lib              28804  3 rt2800usb,rt2800lib,rt2x00usb
sd_mod                 25376  2 
usb_storage            35600  1 
scsi_mod               70488  2 sd_mod,usb_storage
smsc95xx               14168  0 
bcm2835_v4l2           29480  1 

[root@mp-9026ce7f ~]# ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.75  netmask 255.255.255.0  broadcast 192.168.1.255
        ether b8:27:eb:26:ce:7f  txqueuelen 1000  (Ethernet)
        RX packets 670380  bytes 45494321 (43.3 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 641162  bytes 908183341 (866.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 0  (Local Loopback)
        RX packets 275065  bytes 1797627279 (1.6 GiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 275065  bytes 1797627279 (1.6 GiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 00:c0:ca:84:2f:be  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ccrisan, 2015-01-06:
The OS appears to recognize your adapter. What kind of security does your wifi network use?

dkelley810, 2015-01-06:
Network ID (SSID) 2WIRE214 
Authentication type WPA-PSK (TKIP) and WPA2-PSK (AES) 
Network key ********** 
Status On

ccrisan, 2015-01-06:
Everything looks fine to me. Could you also attach the /var/log/wpa_supplicant.log please?

dkelley810, 2015-01-07:
/var/log/wpa_supplicant.log is empty

dkelley810, 2015-01-07:
/data/log/wpa_supplicant.log empty

ccrisan, 2015-01-07:
That's odd. Can you attach a monitor and take a photo of it (or maybe two) that captures the network initialization? More precisely, there are a few init scripts that handle the network initialization; I'm interested in the messages they write to the console.

dkelley810, 2015-01-08:
What if I try a different release of MotionPie? was there a kernel  change a while ago?

ccrisan, 2015-01-09:
Well it has changed a couple of times during the past few months but that's unlikely to be the problem since you get the wlan0 interface. Could you by any chance connect both the wifi adapter and an ethernet cable so I can remotely do some debugging using SSH?

dkelley810, 2015-01-09:
how can i get the pi to recognize hdmi or the composite video. i thought the config.txt file would allow me to enable that.

dkelley810, 2015-01-09:
i was able to get of video of my MotionPi booting. i have it stored on dropbox, that i can share with you.
I also have it setup with with ethernet cable and wifi adaptor if you want to try that way.

Let me know!

ccrisan, 2015-01-11:
Yes, I'd like to do some debugging to see what's going on with your setup. Would you by any chance allow me to connect remotely through SSH to your (ethernet-enabled) PI? You can get in touch with me by email.

dkelley810, 2015-01-11:
Yes that would be great! How do we do that?

Dennis

ccrisan, 2015-01-11:
Ideally you'd drop me an email with the IP address where I can reach your PI through SSH. If you did not change the admin password, I should be able to login with the serial number and run some tests.

ccrisan, 2015-01-11:
For the record, user dkelley810 was kind enough to help me hunt down the problem by providing SSH remote access. The issue turned out to be related to wpa_cli not functioning well with this adapter. It might also be related to the fact that rfkill support is reported by the driver.

Anyways, I decided to use "ip link" instead of "wpa_cli" to determine the status of the wireless connection.

closed 2015-01-11 by ccrisan

dkelley810, 2015-01-13:
The new release works awesome! I got my WiFi adapter working!

Thank You so Much for your help!