ccrisan/motionpie #60: Problem using German Umlaut letters (äöü) in Camera and User names

lastfreenick, 2015-01-08:
Using said letters naming cameras and users returns an error in Motion Pie web Interface. It would be  much appreciated by me and I guess a lot of users from other countries with funny letters if these and possibly other non latin letters could be used to name stuff.

ccrisan, 2015-01-08:
Alright, I'll look into it and see what I can do about non-ASCII characters.

lastfreenick, 2015-01-08:
Now that's a fast response right there ;)

Of course this isn't a very important issue at all, we can get around using Umlaut letters in German but I guess that there are a lot of languages where you can't this easily. 

By the way, is there a way to send you some 'beer money' for your efforts?