ccrisan/motionpie #61: Issue with passwords/adding camera on 20150102

Fetchered, 2015-01-09:
Running on 2x B+ models with standard Pi Cams. Previously had each camera added to the other so I could view both cams on one IP. Since the upgrade I lost one camera on one Pi (likely wasn't detected while it was rebooting). Try to add it using remote motionpie camera option and, while entering the user/pass I notice it says Authentication Failure prior to hitting anything else. As I go to click out of the box (either to OK/Camera etc) it prompts for a Login. I enter both my admin and user, and both appear incorrect. After many attempts, it eventually gives me the prompt: "true" error, has appended "/login/" to my URL. 

At this point I can no longer log in until I close my browser and relaunch, or reboot the Pi. This behavior happens on both Pi's. 

Any help you could provide would be great. Logs to follow if required, attempting to fix an SSH problem right now. 

Thanks!

OS: iOS 8.1 and 8.1.2, OS X Mavericks, OS X Yosemite, Android 4.4.2
Browsers: Firefox, Chrome, Safari
Devices: Galaxy Tab 3, iPad Air, iPhone 5, Mac Mini

ccrisan, 2015-01-09:
Did you upgrade both motionPie instances? Cleared browser cache (this is a quite big update)? Either way, could you please send me both sets of motion.conf, thread-1.conf and motioneye.log? You can replace the password with dummy text but if you have some special passwords (i.e. all digits, spaces or special characters) please let me know.

Fetchered, 2015-01-09:
I did upgrade both, and cleared the caches on all browsers. The motioneye log file is rather large. Any preference for how you want me to send them?

ccrisan, 2015-01-09:
Please email them to me at the email address you'll find in the source code.

Fetchered, 2015-01-09:
Done. Thanks!

closed 2015-01-11 by ccrisan