ccrisan/motionpie #62: motion movies with still images and with jumps on it

boranucleare, 2015-01-09:
Hi all,
I set up positively 3 motiopie systems.
One of these have some strange beaviours.
1) Somethimes  (opten :-) ) when a motion is detected I can see 00:50 of recorded video, after that minutes of blank screen with the time increasing correctly. Which parameters are responbible of that?
2) Sometimes, even toghether with issue 1, there are jumps during the movie; sometimes 5 seconds, sometimes 10 seconds and so on. Randomically. Which parametres I need to check?
3) In another installation Motiopie records the movie, but the only thing I can see is a single picture freezed in the beginning time.Note that the thumbnail in the recording page, has another image on it, so I think the movie is right, but I can see it...  strange..... maybe something related to the file downloading from web interface? something related to my download before the movie is finisced to be recorded?

Let me know if there's something to do
Fabio

dkelley810, 2015-01-11:
I maybe able to add some help. I had an issue with a USB camera attached to my PI. and it would lock and sometimes have part of the image shifted. it was like a slice of it moved to the left or right.
I do not know what camera you are using but maybe try another camera to see if it does the samething. Just my thought, I am kind of new working with MotionPie.

boranucleare, 2015-01-12:
my camera is a CSI flat cable ip camera standard from raspberry 1.3 rev

closed 2015-02-15 by ccrisan