ccrisan/motionpie #63: USB camera - microphone / sound support

DanTheColoradan, 2015-01-10:
I have both the stock Pi camera module and a microsoft camera on there, and so far so good! The auto brightness detection drives me a bit nuts, but that's Motion's fault.

Any way, the microsoft camera has a microphone, but I don't see anything for sound. Is it supported? If not... How hard would it be to implement. I'm thinking of looking over the code base and seeing if I can add it myself. It might be quite a bit over my head though.

ccrisan, 2015-01-10:
To be honest it's a lot of work and I don't have time for it these days. If you think you could implement it, you can fork the project and do your changes. I'd be happy then to review the code and possibly pull it into the main motionPie repo.

DanTheColoradan, 2015-01-10:
K. Sounds good. I'll let you know if I accomplish anything ;-) Btw, I'm not sure if you did this intentionally (if so, awesome work), but Sighthound picks up the Pi automatically. That's pretty dang cool.

Zachu, 2015-04-21:
First of all I just have to say that this project is wonderfully done so far :) Thanks @ccrisan !

Sorry for spamming on this already closed feature request, but I just had to +1 this feature. Support for grabbing the audio would be awesome for me also. I'm mostly building my RasPi for watching what my dog does when I'm not home and audio would be crucial for me too. I don't think I have time or skills to dig into this either, so I guess it's to be buried for a while atleast.

ccrisan, 2015-04-21:
The big problem here is the very foundation of motionPie, motionEye and of course, the motion daemon: mjpeg.

The mjpeg technology has the advantage to be easy to work with (no codecs, simple format). But among its downsides there is this limitation that it cannot carry audio, just jpeg frames.

This being said, either we use a totally separated way of transmitting the audio stream to the browser (flash? html5?) or we rewrite everything from scratch and use a proper video format and codec.

The first one is actually doable, but I could really use some help.

ainesophaur, 2015-07-16:
I would love to assist with this. Which package/repo actually handles the web server for motionpie ?

ccrisan, 2015-07-17:
@ainesophaur [motionEye](https://bitbucket.org/ccrisan/motioneye) is the webserver that takes care of most of the goodies of motionPie. How would you like to proceed about this? What technologies do you have in mind?

ainesophaur, 2015-07-17:
I'm a huge fan of the capabilities of NodeJS and it's ability to handle streams. Unfortunately I'm not as skilled with python as other languages nor do I understand it's full capabilities. I know with nodejs I could take a stream from ALSA and output it directly to an HTTP stream. With monitonEye I could then open the stream. Out of the gate I know there would probably be sync issues between the audio and video but it's not impossible to correct. 

Before finding motionPie I came across [this article](https://www.the-hawkes.de/a-raspberry-pi-powered-baby-monitor-12.html) which used netcat to stream the mic from my webcam through alsa -- I was thinking of outputting the stream to an audio file and streaming it through HTML5

Alexamder, 2015-12-22:
I have done this with arecord.

Just write to lines to the config file.

on_event_end /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf stop %t; ps -ef | grep arecord | grep plughw:%t | grep -v  grep  | awk '{print $2}' | xargs kill -9

on_event_start /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf start %t; arecord -f S16_LE -r 22050 -D plughw:%t /media/motioneye_192_168_178_2_camera_camera/%Y-%m-%d-%H-%M-%S_%t.wav