ccrisan/motionpie #64: Enable motion detection for part of the frame only

SuperSharpShooter, 2015-01-11:
Is there a way to add an option to only enable motion detection for a certain part of the frame?

I'm getting way too many false positives on my camera, as there are trees which sway in the wind at the top of the frame (which sets the camera off), but I am only interested in motion in the lower half of the frame where the driveway is. 

Just using the current sensitivity settings, it's impossible to stop the false positives, while allowing the camera to still be triggered reliably when there is a car or person in the lower frame.

Even if there was a square box of various sizes or positions in which motion would be detected, then the false positives could be cut down drastically.

ccrisan, 2015-01-11:
This is the feature you're looking for: https://bitbucket.org/ccrisan/motioneye/issue/68/image-mask-feature-request. It's going to happen some time in the future.

SuperSharpShooter, 2015-01-11:
Alright, thanks for that. Look forward to it being implemented.

ccrisan, 2015-05-16:
Duplicate of https://github.com/ccrisan/motionpie/issues/64.

closed 2015-05-16 by ccrisan