ccrisan/motionpie #66: SMB configuration

sticilface, 2015-01-11:
I've tried a couple of permutations but I'm not sure how to get the data stored on a network share.  

The fields are

network server : 
share name:
share username
share password
root directory

is the network server just the IP..  i.e. 192.168.1.100 or //192.168.1.100
share name is that the path i.e. //192.168.1.100/pibackup/securitycam/hallway
username and password are self explanatory
root directory - is this the local path to mount it too...  

An example would be good,  I'm not sure if the full path //192.168.1.100/pibackup/securitycam/hallway should be included in the network server field..  just a bit confused as there does not appear to be any detailed documentation yet!  Not knocking it, I'm pretty chuffed how smoothly this has worked right out the box!

and thanks ccrisan for all the replies, and i apologise for posting so much.

ccrisan, 2015-01-11:
Network Server: 192.168.1.100
Share Name: pibackup (always the first thing that comes after the server)
Root Directory: /securitycam/hallway (whatever comes after the share name)

There's a small question mark next to each option giving an example of what the input should look like.

Let me know if it works (or not).

sticilface, 2015-01-11:
worked like a charm!  

thank you very much

closed 2015-01-11 by ccrisan

piecam, 2015-01-14:
HI ccrisan,

First of all, great work that you have worked out this piece of work! I have a few questions for SMB.
I have a NAS SMB share by the name of "Recording" and "Recording2". There are folders inside "Recording", cam1,cam2,cam3. There is folder inside "Recording2" and it's cam3.

I have a Pi Cam Board and it's mounted to //192.168.100.10/Recording
                        870.9G    274.9G    595.9G  32% /da
ta/media/motioneye_192_168_100_10_recording_cam
in thread1.conf.
I have been able to attach another USB logitech c270. When I configured it's SMB mount to same root directory, say "/cam1", the motioneye does't like it and say ERROR: failed to mount smb share "//192.168.100.10/Recording" at "/data/media/motioneye_192_168_100_10_recording_cam".

If I change 2nd cam's root directory to "/cam2". It would be ok and it's working. However, after a reboot, motioneye would complain,  ERROR: failed to mount smb share "//192.168.100.10/Recording" at "/data/media/motioneye_192_168_100_10_recording_cam".

The result that is I don't see any cam image in WEb (motioneye or 8081/8082).

Can you help take a look to see whether we can do the following?
1. let script check whether different cam (thread x.conf) have same SMB share. If it's same, just use the first one as SMB share.
2. If different cams which share same SMB share, just have different "Root Foler", then dont' coplain and just user different folder for each of cam.
3. If each cam has different SMB share, it is working in this one.
4. Can we have a name % to name a captured file with CAM's name+%Y%M%D....etc.?

Maybe I have missed something. Please let me know.

thanks
peng

piecam, 2015-01-14:
The following message is from motioneye's log and it may be saying that "I have first SMB share already, so I can't do the same again.". So, I will close all. Correct?

2015-01-14 08:03:49: INFO: interrupt signal received, shutting down...
2015-01-14 08:03:49: INFO: server stopped
2015-01-14 08:03:53: INFO: motion stopped
2015-01-14 08:03:53: INFO: SMB shares unmounted
mount error(16): Device or resource busy
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-01-14 08:03:55: ERROR: failed to mount smb share "//192.168.100.10/Recording" at "/data/media/motioneye_192_168_100_10_recording_cam"
2015-01-14 08:03:55: INFO: cleanup started
2015-01-14 08:03:55: INFO: wsswitch started
2015-01-14 08:03:55: INFO: thumbnailer started
2015-01-14 08:03:56: INFO: server started

piecam, 2015-01-14:
As a result, no Thumbnail picture or 8081 streaming is working.

thanks

ccrisan, 2015-01-14:
Don't use the same share on two cameras on a motionPie.

piecam, 2015-01-14:
root@mp-9a24c67f ~]# cat /var/log/motion.log 
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-2.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] motion_startup: Motion trunkREV561 Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (EMG)
2015-01-14 08:03:54: ERROR: failed to relay event: [Errno socket error] [Errno 111] Connection refused
2015-01-14 08:03:54: ERROR: failed to relay event: [Errno socket error] [Errno 111] Connection refused

Here's log from motion.log
tks

piecam, 2015-01-14:
HI ccrisan,
that's a quick response!!!! thaks a lot.
So, for each cam we have to use differen't SMB Share? Would future version would help accomodate same SMB share with several cams? That would be nice!

thanks
peng

piecam, 2015-01-14:
Can I make a feature request for this, then?
tks

piecam, 2015-01-14:
HI ccrisan,
I did a workaround by either commenting out network share related lines in thread-2.conf or use custom path with following and it works.
/data/media/motioneye_192_168_100_10_recording_cam/cam2

Hope we have a future version help deal with this .

Many thanks
Peng

bazcurtis, 2016-02-15:
Hi Peng,

Are you saying that the path above is to a SMB share?

Best wishes

Michael