ccrisan/motionpie #68: Unable to switch users

MelGrubb, 2015-01-11:
I have just installed the 1/11/15 image, and clicking the key does not let me switch users. The screen refreshes, but I still don't see any admin interface. This is my first attempt at using the motionPie image, so I could be missing something in the instructions, but I've read through them several times, and I don't see what I could be doing wrong.

I'm using a model B+ with the infrared camera (RS version of the NOIR). The camera is detected without a problem, the web interface is up and running, and I can see the camera image, but I can't make any settings changes.

ccrisan, 2015-01-11:
A known bug, use previous version.

unmanagedtn, 2015-01-12:
So what version is it working in I tried the prev version and it wont work also... 
Any help in this matter? or ETA on a fix?

unmanagedtn, 2015-01-12:
what is a stable release then? I have tried the prev and it still is not working ...

ccrisan, 2015-01-12:
This is the last stable version for now: https://github.com/ccrisan/motionPie/releases/tag/20150102. Hit Ctrl-F5 after opening it in the browser.

closed 2015-02-15 by ccrisan