ccrisan/motionpie #69: MMAL motion integration

lastfreenick, 2015-01-12:
In my neverending quest to make the pi cam usable with motion pi I've stumbled across the Motion MMAL project which should, from what I understand, greatly improve the pi cams overall performance and will probably fix the brightness hanging issue. Any chances thic can be baked into Motionpi? Here's a link In case you havent heard oft it: http://www.raspberrypi.org/forums/viewtopic.php?f=43&t=75240

ccrisan, 2015-01-12:
Yeap, this could be a good idea. I never had the chance to test this motion-mmal though and see if it brings a real improvement over the mainstream motion with v4l2. I'd be happy if someone could try and integrate this as a custom BuildRoot package for motionPie.

I will then make sure both versions are included in the OS and I could patch motionEye to start motion-mmal if and only if one  CSI camera is configured.

lastfreenick, 2015-01-15:
I'd just wish, I could help out. These pi cam issues are driving me nuts. I've considered just using plain motion mmal on a clean raspian install but I really love the sleekness and simplicity of your interface. I guess replacing the motion build in motion pi with motion mmal won't be something a noob like me can handle. The pi cam, especially the noir version, is a great piece of hardware, cheap and tiny, great for home built security cams.

lastfreenick, 2015-01-15:
I just found what I think is another driver for the pi cam, that is reported to play well with motion. Please have a look and excuse my noobishnes in case this is rubbish: http://www.linux-projects.org/modules/news/article.php?storyid=150

ccrisan, 2015-01-16:
This is an on v4l2 compatibility module for the mmal camera. It's now obsolete as the driver is included with the RPi kernel.

lastfreenick, 2015-01-18:
I've been playing around with manipulating the camera directly via v4l2-ctl and have found that I can get some pretty nice resuts from the camera, especially in low light situations, turning auto exposiure off, raising iso sensitivitiy and setting exposure time to max. I also feel that changing the auto exposure metering from average to something else might possibly fix the brightness detection (I've only started playing around yesterday evening, so this needs at least a full day/night cycle to test).

Anyway, having found that one can change advanced parameters at will, I would like to ask you for a way to permanently set these settings because rebooting the pi overrides all my changes. (A more advanced  camera settings menu in motion pi would be super neat) Furthermore, is there a way to change these parameters depending on the daytime? the cron command doesn't seem to be included in motionpi.

ccrisan, 2015-01-19:
Adding support for cron wouldn't be such a bad idea. However adding UI for this kind of settings is out of the scope of motionEye/motionPie.

Regarding the original request of this issue, I managed to cross-compile motion-mmal and it will be included in the next release, along with the vanilla motion. Users wishing to toy with it will have to rename these files around a bit.

I tried this motion-mmal with a RPi CSI camera and I saw no sensible improvement. There might be some particular cases where it can bring speedups but I don't think it's worth making an automated choice of motion binary based on current camera.

lastfreenick, 2015-01-19:
A massive THANK YOU for your outstanding support!

lastfreenick, 2015-01-20:
I've had the chance to test the different exposure metering modes as well as other exposure related settings in v4l2-ctl and I'm now absolutely sure automatic exposure is bugged/broken in v4l2 when used with the pi cam. Sonetimes the camera readjusts exposure for a while but then it just gets stuck at a certain value, leading to overexposed or underexposed images. If motion mmal fixes this issue, you could maybe reconsider your idea to automatically load mmal when motionpi is used with a pi cam. For now I've resorted to scrips that reset motion after a fixed period and set the cam to a night setting in the evening, but this is quite ugly and rough stuff.

ccrisan, 2015-01-20:
You can download the motion-mmal binary compiled for motionPie from [here](http://www.calincrisan.eu/public/motion-mmal). Just replace your motion binary with this one and you should be able to run some tests and draw some conclusions.

lastfreenick, 2015-01-20:
Once again thanks a lot. Just to get it right, by replacing I guess you mean the file called motion* in /usr/bin and I guess I will have to rename the file provided to motion*

ccrisan, 2015-01-20:
Yes, you need to rename it. Let me know of your findings.

lastfreenick, 2015-01-20:
Got the motion binary provided by you up and running. After replacing the old motion binary I forgot to make the new one executable which resulted in an old fashioned boot loop, luckily the system was up long enough to make the file executable. After correcting my mistake, the new binary runs like a charm. It's dark now here in Germany so I will have to wait for tomorrow to see if this binary will fix exposure hanging.

The thing that confuses me a bit is that I can still manipulate camera settings via v4l2-ctl. The way I understood this whole motion mmal thing was that motion mmal uses it's own way to interact with the camera module instead via the v4l2 driver. But then again, I'm overly happy to having achieved to replace the binary at all. Linux shell is a whole new world for me filled with mystery ;)

lastfreenick, 2015-01-21:
I kept a close eye on the camera picture this morning and even though the brightness did in fact adjust, I ended up with a greenish picture which also sometimes happened with the original binary:

[image]

lastfreenick, 2015-01-21:
I will check the config settings when I get home, maybe disabling contrast and saturation in the config will help, it sometimes did so with the original binary

lastfreenick, 2015-01-21:
contrast saturation and brightness were already set to 0 so there is one last resort for me before I will go either nuts or have to check if my camera is faulty:

From the mmal example conf:

# Switch this setting to "on" to use the still image mode of the Pi's camera
# instead of video. This gives a wider field of view, but requires
# a much slower frame-rate to achieve exposure stability
# (e.g. 0.25 fps or slower). You can use the minimum_frame_time
# parameter above to achieve this

mmalcam_use_still off

Will try and report.

lastfreenick, 2015-01-22:
Finally!

Setting the cam to mmal stills mode solved all my hanging issues. The cam went down to a very good low light setting last light and went back to correct exposure this morning. No green effect, no overexposure. I guess what the cam does now really is to take continous snapshots so the exposure metering is done every frame and therefore the hanging is resolved.
The only downside of using stills mode is that I'm limited to 2-3 fps but that is ok with me I prefer high res over fluency. 
For the sake of scientific verification I've put together another motion pi cam from parts I've bought for another Cam I'm currently building. I'm still waiting for a ir switch filter I've ordered from china but I put the remaining parts together and turned it on to see if it also has brightness issues. Will report my findings.

lastfreenick, 2015-01-23:
Motion mmal in stills mode is still going strong, delivering a well lit picture in low light situations, better actually than I have ever been able to achieve with the regular motion (messing with all kinds of settings). Meanwhile on my other motionpi this is what i get after one night cycle(above mmal stills, below regular motion, second camera pointing out of a window):
[image]
[image]
[image]

lastfreenick, 2015-01-24:
For the sake of completeness, I installed mmal on my second motionpi and set it to stills mode: 
[image]
Again much improved low light performance and no more 'green screen'. If it was possible to enable some kind of 'stills mode' in regular motion  I guess this would do the job as well. Looking at performance mmal looks slightly better to me, I monitored cpu load via the android app raspi check, running mmal seems to lead to slightly lower cpu load. 

So my conclusion is mmal is indeed useful for it's stills mode. I guess most users of security cams perfer low light performance over a high framerate, which itself is limited by the pi's power.

ccrisan, 2015-02-15:
motion-mmal binary is now available on the image, but needs to be manually renamed, for now.

closed 2015-02-15 by ccrisan

piecam, 2015-02-18:
HI ccrisan/lastfreenick,

Can you help write a quick guide of steps to change to use mmal ? I have experienced simialiar issue with resultiion over 1280x720.

thanks in advance