ccrisan/motionpie #70: Streaming URL requesting login

snagglewest, 2015-01-14:


snagglewest, 2015-01-14:
The Streaming URL in 20150112 is acting a bit strange.  From an internal network it works fine with a url of http://theIPadress:8081 but when trying to access the streaming url from an outside network it goes directly to the login even after giving the proper login.  It only displays the main page with the image trying to load, but never does. The port forwarding is correct as it works fine in 20141123.  The Snapshot URL works as it's supposed to though.

ccrisan, 2015-01-14:
On 8081 there's another server that listens - it's the motion daemon in fact. If you get any kind of login prompt, then there might be something wrong with the forwarding. Make sure you're not forwarding 8081 to 80. Also you shouldn't access the streaming URL directly from a browser.

snagglewest, 2015-01-14:
Sorry, Not sure what was going on but  wiping the browser cache on and a reboot of the motonpie and the router fixed it.  I was only using the browser to test the connection. I'm experimenting with using motionpie in conjunction with zoneminder for the ability to monitor different camera types from various networks and the setting alarm zones.  I'm happy to report motionpie works very well with zoneminder.  Thanks!

ccrisan, 2015-01-14:
I'm glad to hear that!

closed 2015-01-14 by ccrisan

rafi1975, 2015-07-28:
Hi,

What setting are you using for the camera in ZoneMinder?
I can only see a blue screen with proper date and time from motionpie.

Thanks,

Rafal