ccrisan/motionpie #71: Raspberry pi A+ crashing when accessing GUI

sticilface, 2015-01-14:
Is there a way to identify what the issue is that is causing it to reboot? 

There does not seem to be anything in var/log/messages..

A

ccrisan, 2015-01-14:
Power supply issues? Also make sure the configured connection actually works. There's a "software" watchdog that will reboot your PI when the network connection (either wifi or ethernet) is down for 20 seconds, if enabled.

sticilface, 2015-01-14:
I'm fairly confident that it's not those. The power supply is a decent usb one that I've been using for ages with a b+ sans problem. 

The wifi is working as I can ssh just fine. It seems to be just once I access the web gui. I get about 5 seconds of video stream, then it stops, and reboots.

ccrisan, 2015-01-14:
Try using writeimage.sh -w and see if it still happens. If it does, I'd like to see the contents of the various /var/log/* files (an archive of /var/log would be great).

sticilface, 2015-01-14:
Will do it this evening!

sticilface, 2015-01-14:
ok, 

I've narrowed the issue down.  if i change the resolution to anything apart from the default... of 320x something... i.e. really small.. it just hangs and goes round in a reboot loop...  even though it is an a+..does the loss of half the ram affect it that much? 

I've done firmware update... still does it..

this is the command I'm running 

sudo /Users/amelvin/Downloads/writeimage.sh -w -d /dev/disk2 -i /Users/amelvin/Downloads/motionPie-20150112.img -l -n wifiuidd:abcde

it works if i take out the -w...  if i put it in there or anywhere else.. then the image never boots... by the terminal i get... decompressing image.. booting...  message...

A

ccrisan, 2015-01-15:
There might be an issue with writeimage.sh on OSX - I cannot test this. Nevertheless, A+ should be enough and memory shouldn't be a problem. What camera do you use? If it's an USB camera, can you try to use another camera model and see if the problem persists?

sticilface, 2015-01-15:
The camera is the pi camera, pinoir...  csi i believe

ccrisan, 2015-01-15:
Try to reduce the CPU frequency; the overclocking setting I chose for motionPie might be too much for your unit. Use [these instructions](https://github.com/ccrisan/motionPie#tweaks) to make changes to your /boot/config.txt and set arm_freq=700.

ccrisan, 2015-02-15:
The default overclock frequency has been changed to 800 now. Let's hope the A+ problems are now gone.

closed 2015-02-15 by ccrisan