ccrisan/motionpie #72: Ralink WLAN USB device causing reboot?

lubomski, 2015-01-15:
great job with motionPie! It's exactly what I have been looking for...

I've managed to setup the camera and have been using the system via Ethernet. Upon connecting a Ralink rt2800 sub adapter and setting my SSID and ow for my network, the Pi reboots. It then gets to the point during reboot where the onscreen message states:

Starting wpa_supplicant [9.543667] ieee80211 phy0: rt2x00lib_request_firmware: info - loading firmware file 'rt2870.bin'

It then seems to hang for a few moments, then displaying a message along the lines of 'device busy' which then prompts a sigterm reboot...

I've tested the adapter using raspbian os and it works just fine. 

Any ideas?

lubomski, 2015-01-16:
here's a potato-quality image of the boot (at the hang point):

[image]

ccrisan, 2015-01-16:
Does it work with other types of wireless adapter? What network security do you use? Do you have special characters (including spaces) in your SSID or PSK?

lubomski, 2015-01-16:
I unfortunately don't have another wireless USB adapter to test with; I use WPA2-Personal on my network. PSK is alphanumeric and SSID has spaces in it.

ccrisan, 2015-01-16:
The security is alright; I believe it's the space in the SSID that's problematic - it's been reported a couple of times.

lubomski, 2015-01-16:
I'll give it a whirl when i'm back at home and see if changing the SSID resolves it.

lubomski, 2015-01-17:
I logged into the Pi using PuTTY and added my SSID (with spaces) and PSK into the /etc/data/wpa_supplicant.conf and the Wifi connection is now working...

ccrisan, 2015-01-17:
Can you tell the difference between the two (working and non-working) wpa_supplicant.conf files? I would like to see these files, but you probably don't like sharing your network passwords.

lubomski, 2015-01-17:
Non working had following (recalling this from memory):
{
   ssid_scan=1
}

Working:

{
   ssid="this is my ssid"
   psk="0123abc45DEF"
}

It's almost as if the GUI didn't / couldn't write the keys to the wpa_supplicant.conf

ccrisan, 2015-01-17:
Hmm that's odd. I was thinking more like special characters (unicode space replacements or so). I wasn't expecting the file to be empty. Are you able to reproduce the problem? And by the way, are we talking about /data/etc/wpa_supplicant.conf, right? (as opposed to /etc/wpa_supplicant.conf).

lubomski, 2015-01-17:
Yes we are. I have a spare SD card; I'll follow the installation from
scratch again and see if it is reproducible...

lubomski, 2015-01-17:
okay; so vanilla installation (cabled connection), ssh'd in and contents of /data/etc/wpa_supplicant.conf are as follows:

    update_config=1
    ctrl_interface=/var/run/wpa_supplicant

    network={
        scan_ssid=1
     }
Powered down, connected USB wifi adapter, powered back up, no reboot issue. wpa_supplicant.conf is as above.

Opened GUI, enabled advanced controls, entered SSID and password into interface, rebooted... 

On reboot, connection was established to wifi network and upon SSHing back in, wpa_supplicant has been updated with the correct details! 

    update_config=1
    ctrl_interface=/var/run/wpa_supplicant.conf
    
    network={
        scan_ssid=1
        psk="0123abc45DEF"
        ssid="this is my ssid"
    }

Apologies, it seems I may have a 'ghost in the machine'...

ccrisan, 2015-01-17:
Given that you're not the only one reporting this issue (or similar), I believe it's a hard-to-reproduce but a real problem.

I will leave the issue open for now.

lubomski, 2015-01-17:
I'll try another couple of scenarios and see if I can trigger the issue... I'll post back if I can replicate the problem

closed 2015-02-28 by ccrisan