ccrisan/motionpie #73: supported usb wifi adapters

nazaretvaz, 2015-01-16:
hi

Could you please provide the list of supported usb wifi adapters. I have a D-Link DWA121 which is not detected.

thanks.

ccrisan, 2015-01-16:
I don't know how to compile such a list. Does it work with Raspbian?

lubomski, 2015-01-17:
Supported: PiHut USB 802.11b/g/n adapter (Ralink chipset) http://thepihut.com/collections/raspberry-pi-accessories/products/usb-wifi-adapter-for-the-raspberry-pi)

nazaretvaz, 2015-01-23:
[image]

D-Link DWA121 worked and able to connect to wifi but few seconds after the pi is powered on it crashes. please find the attached screenshot with the error message displayed.

let me know if you can help.

ccrisan, 2015-01-23:
That "surpriseremoved" message indicates the usb disconnected the device somehow. Check your power supply.

nazaretvaz, 2015-01-24:
Tested on Rasbian, no issues works fine. any idea? i dont have a powered usb hub to confirm the power issue.

Tjanko, 2015-02-22:
Hi there, 

Apparently there also is an issue with Realtek RT9182CU - an attempt is made to run wpa_supplicant and after a while system reboots (looping). 
According to this https://github.com/robertalks/rt8192cu - the only remedy is to use original Realtek driver. This requires though as I believe recompling the image ?
I am using very popular nano and regular size Edimax USB dongle. Both use the same chipset, and despite number of attempts I can not make it work with motionPie. They work flawlessly with Raspbian.
Entering WiFi credentials via motionPi web interface does not even save them in wpa_supplicant.conf.
I have made changes manually via shell, same result - rebooting with no WiFi connection established.

Any help appreciated.

ccrisan, 2015-02-22:
The driver is the "original Realtek driver"; motionPie uses  the raspberry pi patched kernel, the same as the one used in Raspbian. I don't know what to say about the system not recognizing it, but I'm willing to help you debug the problem over a remote debugging session (SSH port forwarding, through wired connection, teamviewer or whatever).

Tjanko, 2015-02-22:
Thanks for prompt reply. 
Strange I tried with other access point and to my surprise... it worked. I thought that the issue might be related to the dash in SSID identifier. I changed then SSID for the access point that I did not managed to make work with motionPie. No success though, I must investigate deeper. Thanks for for offering me a hand. I will ask for your help if I do not manage to force the first AP to work with motionPie. 
Best

closed 2015-02-28 by ccrisan

karma007, 2015-03-01:
Hello, i could not make it work with my Dlink Hub, motion pie was always rebooting (Rasp B).
when the picam is not connected, no reboot :)
I had to use direct supply power to make it work. Is it because of the picam, because i used raspbian without cam with this hub, and a lot of usb devices pugged ?

ccrisan, 2015-03-01:
It is probably related to the power supply and the fact that motionPie is (was) overclocked by default. The kernel and drivers are the same as in Raspbian.