ccrisan/motionpie #74: Anybody have experience using Trendnet cameras with motionPie?

IAmNotARobot, 2015-01-17:
Hi, has anyone tried to setup motionPie with any Trendnet cameras? In particular, I am trying to set up the TV-IP343PI camera < http://www.trendnet.com/products/proddetail.asp?prod=180_TV-IP343PI > so motionPie will able to get a live video from it.

I know for a fact that this camera does have a web viewing application, an rtsp server, as well a UPnP, Multicast, Onvif,, Bonjour and LLTD. It also supports outputting in the MJPG file format. But I can't figure out how to set it up with Motion Pie (if its even possible) since the rtsp isn't supported by motionPie at the moment.

 Any suggestions of things to try?
Thanks!

ccrisan, 2015-08-09:
I'm currently working on implementing support for RTSP netcams in motionEye/motionPie. The biggest problem is that, as I don't own a RTSP-enabled camera, I'm using VLC to simulate one.

Things are working fine with VLC, but I would like to test it with a real RTSP camera. Would you be able to configure your RTSP camera for remote access, so I can use it for testing?

martymoose, 2015-08-09:
I have a spare  ip cam i can set up for you

ccrisan, 2015-08-09:
@martymoose that would be great. Ideally you should set it temporarily as DMZ in your router, unless you have some other advanced network configuration.

martymoose, 2015-08-09:
Mate on night shift do ill get done tommirow  about 12hr from now

ccrisan, 2015-08-09:
Sure, no rush. Just let me know when you have set it up. Thanks very much for your help.

martymoose, 2015-08-10:
ok have a cheap hunt brand with rstp enabled and tested

email me your email address to martyATmartymandy.com

*Martin Thompson*

*Po Box 1415*
*AYR   QLD    4807*

*M. 0466 005866*
*E.  marty@martymandy.com
<https://mail.google.com/mail/u/0/?view=cm&fs=1&tf=1&to=marty@martymandy.com>*
*W. **martymandy.com* <http://martymandy.com>

Check the weather at my place martymandy.com

ccrisan, 2015-08-11:
@martymoose, I'm ccrisan at gmail.

martymoose, 2015-08-11:
Still need to test? If so ill turn on when i get home

IAmNotARobot, 2015-08-12:
@ccrisan I'm sorry I disappoint. A lot has happened in the last few months, and we're actually renting out our house now so I no longer have access to those cameras. But good luck to you in your development! I know a lot of home security cameras use RSTP so this will definitely open up a lot of doors for motionPie.

If I may make a suggestion though, Logitech had a line of awesome security cameras that they dropped support for. If you do end up getting the RSTP connection working well, you may want to drop a note on the Logitech forum about it. Many people will probably eventually want to drop their old Logitech Alert monitoring software because new operating systems may eventually make it incompatible. Some people are already doing this and switching over to programs like Blue Iris, but some (like me—if I still had access to my cameras, two of which were Logitechs) might be interested in the ultra low power consumption of a RaspberryPi based system. So this may be a good audience to look in to. But of course if making motionPie compatible with these cameras will require a lot of work, don't waste your effort since most people are still satisfied—for the most part—with the default monitoring program.

If you are interested, however. Check out the pictures on [this site](http://forums.logitech.com/t5/Alert-Security-Systems/Blue-Iris-The-Best-Alternative-Software-For-the-Alert-System/td-p/498968) about how to set up a Logitech Alert camera via RSTP. They show the path, username, and resolution you need for the camera.

Thanks for adding this feature by the way! I hope it gets motionPie even more recognition in the camera world! Good luck with everything!

ccrisan, 2015-08-12:
Thanks for all the info, @IAmNotARobot. There are a lot of netcam improvements and fixes coming up with the next release.

IAmNotARobot, 2015-08-12:
@ccrisan Nice!