ccrisan/motionpie #75: configuration not applied after upgrading new version 20150112

piecam, 2015-01-17:
HI,
I just upgraded to this new version on two pi b+ with pi cam on both. I looked like that all configuration setting are there in GUI. However, I have to turn on/off each part on Still Image, Motion Detectoion and Resolution to get it to work again captureing motion.

I have seen no log errors. Not sure it's only for my two Pi B+ unit. Any one else seen this?

tks
peng

ccrisan, 2015-01-17:
Does it also happen after you refresh the page?

piecam, 2015-01-17:
Basically pi got software upgraded and rebooted. It's evening time around 8:30 ish. It's dark outsite. I checked the result. It looks ok. As it's dark outside, I didn't go to test motion. This morning, I booted my computer and opened browser into it. I noticed that I don't have file on SMB when I did a walk. I started investigating that leads to this finding.
So, yes, it could be consided a browser refresh. 
thanks
peng

piecam, 2015-01-17:
I tried to adjust Motion Threshold. It was set for 5% and it worked well. Then, I modified to 10% and it didn't capture anything. Then, I modified down to 7%, nothing. Then, 6% nothin. Then back to 5% nothing.

I was walking out side to test.

In the previous version, I can modify this up and down. When it's back to 5%, it works. Is there possibility that some code change might have affectd this?

thakns
peng

ccrisan, 2015-01-17:
That code hasn't been changed in a while.

piecam, 2015-01-17:
I will test some more. thanks

closed 2015-02-15 by ccrisan