ccrisan/motionpie #76: cpu, on motionpie vs minidibian with motioneye

piecam, 2015-01-17:
HI,
I have a typical usage on my pi b+ with pi cam about 60%-80%, with no web gui access. It has resolution 1600x1200, frame rate 2/sec, streaming 5/sec. CPU goes to 90%+ with GUI Access or higher resolution like 2592x1944.

Are these typical? anyone else can share their cpu usage? tks

I am going to compare CPU with motionpie image vs minidibian with motion eye. Has anyone done this before?
tks

piecam, 2015-01-18:
it looks like CPU is directly related to resolution setting. With 1600x1200, I have Motion process using cpu constantly around 90%. With resolution 1440x1024, it's about 80%. If going lower to 1280x720, it would be 50%.

Is there anyway to use high resolutin while still keeping Motion's process CPU low?

thanks

ccrisan, 2015-01-18:
No, there's no way to do that. The bigger the resolution, the more pixels to process and thus the higher  CPU usage.

piecam, 2015-01-18:
HI CCrisan,
I was wondering whether you can provide some advice for my following thoughts.

I do have a spare PC with linux and more cpu/ram. If I want totally turn my Pi b+ and pi cam into a pure IP CAM with High resolution (2592x14xx, or 1600x1200, etc) with more frame rate say (5-15/sec). Then, as long as those high resolution stills befing feeded to this more power pc , I can use your motion+motioneye, or even zonefinder to centralize and process it. In this scenario, what's best programs available? I know there are some on the net. I thought that You may have better insights that I can leverage.

Hope that will keep pi in not very high cpu and being able to feed high quality stills or mjpeg or h264 out.

thanks in advance
peng

SuperOok, 2015-01-18:
Does the load also linearly scale with fps? So is there some maximum, let's say , "megapixels per second" value that the pie can handle? With such a value one could calculate the best setting for resolution and fps. I would assume this max value would be higher if no motion detection is done by the pi, right? And even if you don't use the web gui but only streaming? Any experience with the pinoir cam?

IAmNotARobot, 2015-01-18:
Hi @piecam ,
Have you tried the overclocking settings for the Pi? Ccrisan has various different settings defined in the installation script on the [motionPie GitHub page](https://github.com/ccrisan/motionPie):
> `[-o none|modest|medium|high|turbo]` - overclocks the PI according to a preset (e.g. -o high)
> *default* - arm=900Mhz, core=500Mhz, sdram=500MHz, ov=6
> `none` - arm=700Mhz, core=250Mhz, sdram=400MHz, ov=0
> `modest` - arm=800Mhz, core=250Mhz, sdram=400MHz, ov=0
> `medium` - arm=900Mhz, core=250Mhz, sdram=450MHz, ov=2
> `high` - arm=950Mhz, core=250Mhz, sdram=450MHz, ov=6
> `turbo` - arm=1000Mhz, core=500Mhz, sdram=600MHz, ov=6

Try to set it to `turbo`.  For example:

    ./writeimage.sh -d /dev/disk1 -i motionPie-20141207.img -o turbo

While this won't make you cpu usage go down to zero, it should bring the usage down at least a little bit. But, I think you don't need to be concerned the cpu is at high load. The RaspberryPi uses very little electricity to begin with, so even at full load it will probably use a lot less power that your other computer. I believe it's in the ballpark of 3 Watts. Compared to a lightbulb (13-100 Watts), it's pretty small.

If you're worried about the RaspberryPi heating up and breaking. I think that isn't really a big concern either, since it normally runs pretty cool anyway, and even if it does break, these are pretty inexpensive to replace. But if you are concerned, check out some on-chip [heat sinks](http://www.frozencpu.com/cat/l2/g40/c16/list/p1/Air_Cooling-Chipset_HeatsinksCoolers.html) from FrozenCPU.com. [This one](http://www.frozencpu.com/products/11755/vid-154/Akasa_Cool_Blue_Anodized_Univeral_Chipset_Cooler_-_40mm_Blue_LED_Fan_AK-210.html?tl=g40c16), for example, looks pretty cool (if you can get it to fit on the board). Even a really simple fanless option should still be able to help a bit. I personally used ones like [these](http://www.frozencpu.com/products/5518/vid-82/Enzotech_Forged_Copper_VGA_Memory_Heatsink_Multipack_-_ATI_and_nVidia_-_14mm_x_14mm_x_14mm_BMR-C1.html?tl=g40c16) for a home project once (they're nice since they come with some 3M thermal tape on one side so you don't need any thermal epoxy or grease to attach them).

But ultimately, heat an cpu load shouldn't make that much difference for this application since as long as the load is below 100%, you shouldn't see much of a performance loss. I wouldn't really worry about it unless you're having skipped/dropped frames. If you do, however, you could try to lower the frames-per-second rate of your camera. For a home security application, 5 fps should be enough to capture people walking/running and neighborhood-speed cars (20 mph @ 20 feet away  =  about 5 frames of the car).

Hope this helps!

snagglewest, 2015-01-19:
I think you might have some unrealistic expectations... the Pi is only a 700mhz ARM11.  Expecting it to process for motion, time stamp, resize, etc AND stream 2592x1944 (over 5 MILLION pixels in EACH image) at anything nearing real time is out of the realm of realistic.  Add in wireless and... well...  Now, if you offload all the extra processing to something like Zoneminder and just send raw 1280X720 it works fine over wireless.  I have ZM set to do all the processing so to speak, flip the image, time stamp, motion detection etc which significantly improved the responsiveness.  On the other hand, a functional wireless IP cam with 1280X720 (or better) resolution for less than $100 US is a heck of a deal!

IAmNotARobot, 2015-01-19:
Well, I wouldn't go so far as to call it unrealistic, but yes, Snagglewest does have a point. There is only so much that a small computer can do. That being said, however, are humans not smarter than computers? We are in fact. And, with a little effort, creativity, and flexibility; it is entirely possible to make the RaspberryPi perform beautifully not only at 2592x1944 but even higher.

The key here is that there are lot's of factors that we can play with to optimize the device to suit our needs. More specifically, it is possible to handle very high resolutions at acceptable processor loads by compensating with either decreased frame rate, lowered image compression quality, or reduced color depth, or even all three.

[Here](http://hornbeam.cs.ucl.ac.uk/hcs/publications/Keval+Sasse_The%20impact%20of%20frame%20rates%20on%20user%20performance%20in%20a%20CCTV%20detection%20task_ACMMM2008.pdf) is an interesting paper about some research regarding the optimal frame rate for CCTV cameras. Check out the Abstract and Conclusion sections.

In terms of image compression and downsampling, there is so much you can do. Some of the more interesting ideas, as far as I can recall right now, include reducing the [image quality](http://en.wikipedia.org/wiki/Lossy_compression) in low interest areas; performing motion detection on very-undersampled images (since most of the time we only care about large moving object—i.e. people and cars but not birds or insects); and reducing image quality on everything except the moving object. Speaking of image quality, check out [this image](http://en.wikipedia.org/wiki/JPEG#mediaviewer/File:Felis_silvestris_silvestris_small_gradual_decrease_of_quality.png). We see here that the quality is really low on the left and pretty high on the right, but in the middle of the picture (between the cat's eyes), our we almost can't tell that it's a lower quality than on the right until we zoom in.

(PS. I'm not sure if these suggestions about image quality will void a video's credibility in court, so if you need to use this camera in a high-cost-of-failure application like a jewelry store, I would actually recommend getting a professional surveillance system instead.)

Lastly, [color depth](http://en.wikipedia.org/wiki/Color_depth) is one of your biggest friends. You can easily save the device some significant processing load by filming in lower color depth. If color is not very important to you, try recording in 8-bit black-and-white (even 4-bit *might* be acceptable in some applications). This should make a fairly nice reduction in you cpu usage.

So really, it's in your hands. If you wish to get unusually high resolution footage with a RaspberryPi, you most certainly can, and there are may ways to achieve the desired effect. But in the end, such features are not likely to be available in most free (and even commercial) security programs, and your best bet is probably to get the source files from Ccrisan's the Git Repository and modify them as needed for your specific application.

piecam, 2015-01-23:
wonderful post and reading from you guys!
If there's a fund raising to get a team of experts on  this program, I am on it. tks

sh2515, 2015-01-25:
If you look at the python API there is a way to have 3 streams from the picamera. 1 can have a lower resolution, so in theory you can analyse that for motion and capture the higher resolution when motion detected. However it would be a completely different project to motionpie.

lastfreenick, 2015-01-26:
I've been messing around with motion_mmal and motionpi. motion_mmal is a module for Motion that reads frames directly from the Pi camera via the MMAL API. Calin was kind enough to provide me with a binary of motion_mmal to replace the regular motion binary provided with motionpi. 
One very neat feature of motion_mmal is that it provides a secondary image buffer:

Secondary Image Buffer
You can now specify a secondary image buffer with a scaled-up resolution from the primary image. This means you can perform motion detection on a low-res primary image (thus reducing the CPU load) but have a higher-resolution, better quality output from the secondary image.

I've not tried to set this up yet, because I'm messing with motion_mmal mainly because motionpi has some serious auto exposure issues with the pi cam), but I guess using a secondary image with reduced resolution could seriously improve performance of motion detection when using high resolutions for the video output.

ccrisan, 2015-02-15:
There will soon be an UI-based way to change the motion binary. I will close this issue for now.

closed 2015-02-15 by ccrisan