ccrisan/motionpie #78: Wifi will not work

Pixelope, 2015-01-17:
Hi
I have been able to set up motion pie through Ethernet and it works great, but for the life of me I cannot get wifi to work. I've looked at all the issues reported here and every solution doesn't work. My wifi uses wpa, it's uppercase, the ssid is correct (no spaces, correct case). I've tried all possible permutations of having the wifi adapter in/out before/after removing Ethernet etc, I literally cannot get this to work how it claims it should. I tried setting up motion through Debian as an alternative but I have no linux experience so that was just a disaster. My wifi adapter is an edimax EW-7811UN which is compatible with the raspberry pi and it works perfectly in my windows pc. Every time I try to use it for the motionpie it either keeps rebooting or if I get to the part where it says login on the screen, it doesn't show up as connected to my network.
I have a powered usb hub (D-Link DUB-H4/B) powering my pi which provides more than enough power to the pi. And the pi and camera work fine as I'm able to use it through Ethernet perfectly.

If anyone can provide some insight as to why wifi doesn't work I'd be eternally grateful.

Many thanks,

Pixelope.

ccrisan, 2015-01-17:
Can you post the contents of /data/etc/wpa_supplicant.conf? Also the /data/log/wpa_supplicant.conf can be useful to identify the problem.

rpilover, 2015-01-18:
I had the same problem so I inserted my SD card in to my laptop running linux. I then edited static_ip.conf using sudo nano ..path/to/sd/card/etc/static_ip.conf adding the details below.

static_ip="192.168.1.50"
static_gw="192.168.1.1"
static_dns="192.168.1.1"

I then saved the file Ctrl+X and inserted the card in to the raspberry, booted up and it worked fine.

Pixelope, 2015-01-19:
thanks for the swift response, I've been having trouble reading the linux partitions on my windows pc.
When i mount the disk, it comes up with two images as part of the sd card, the results of the files are as follows:

disk1s2 - no wpa log file there.
------------
disk1s2 - etc/wpa_supplicant.conf

update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
}

------------

disk1s3 - etc/wpa_supplicant.conf

update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
    psk="0F78F8883B"
    ssid="PlusnetWireless06327D"
}

------------

disk1s3 - log/wpa_supplicant.log
ioctl[SIOCSIWAP]: Operation not permitted

------------

Not sure what this log file means, I'm wondering if i need to manually add my wireless details to the first spa_supplicant.conf file as that for some reason is empty.

ccrisan, 2015-01-19:
Your files seem alright to me. The first wpa_supplicant.conf is the default one, used to create the second one if missing. The Operation not permitted error is also common for the adapter you are using (I myself use a lot of 7811Un adapters these days with my PIs). Did you check to see if your setup works with Raspbian?

Pixelope, 2015-01-19:
Thanks, yeah I have been able to get it all working with raspbian (desktop gui, wireless working). I then attempted a second install using raspbian and motion, but that was incredibly complicated for me to set up that's when I discovered your software.
I have yet to try rpilover's solution, so maybe that might resolve it.

ccrisan, 2015-01-19:
rpilover's solution will solve the issue in case of a DHCP problem. WiFi on the other hand needs to work a priori. Does your network usually take a long time to establish a connection? motionPie waits for at most 20 seconds.

Pixelope, 2015-01-19:
I'm not entirely sure how to check this accurately, but if I connect a new device (tablet for example) it connects and is able to access the network within seconds, probably no more than 5. perhaps I should try a different router?

ccrisan, 2015-01-19:
Changing the router is not a solution in my opinion. We should be able to fix this, or at least identify the root cause. I can ssh into your motionPie (connected through ethernet) if you trust me enough and if you're able to set up port forwarding. Then I should be able to find out what the problem is.

shawnsnider, 2015-01-22:
I also have a similar issue.  I'm on a Linksys 1900AC dual band router.  When connected via 2.5GHz everything is fine.  When I change the wireless settings to connect via the 5GHz network, the pi goes into a reboot loop.  If I remove the wireless dongle, it will boot with ethernet fine allowing me to revert settings.  Is the 5GHz band possible with motionPie?

ccrisan, 2015-01-22:
There is no such restriction so it should work just fine with 5GHz. Your adapter might not support 5Ghz band though.

snyder250, 2015-01-23:
Motion Pi is awsome!!! Everything works great, however I can't get the WiFi working. I searched for the "static_ip.conf" file and cant find one. I tried making on with nano and was denied permission.I read in one of the tutorials that WEP and WPA is not supported. so how can I set up a static IP since the "static_ip.conf" is missing and will it work with WPA2? Other than this problem I love it. GOOD JOB Guys!!!

ccrisan, 2015-01-23:
@snyder250 read [this](https://github.com/ccrisan/motionPie#tweaks) and [this](https://github.com/ccrisan/motionPie#manually-editing-network-configuration) if you want to tweak your system. "nano /data/etc/static_ip.conf" should not give you permission denied.

snyder250, 2015-01-23:
I can't find  a file static_ip.conf in /data/etc/ or /etc/ when i try to make the file with nano and save it i get a permission denied. I did get the WiFi working when i changed to WPA2 with DHCP, but i would like to assign a static IP.

ccrisan, 2015-01-23:
You will not find that file unless you used writeimage.sh to set static ip configuration. Otherwise you need to create it with nano. Permission denied in /data/etc doesn't make sense to me.

snyder250, 2015-01-23:
this is a cut and past of the error message I get when I try to save the new file

  GNU nano 2.3.6                                  File: static_ip.conf                                                                  Modified

static_ip="192.168.1.10/24"
static_gw="192.168.1.1"
static_dns="8.8.8.8"

                                               [ Error writing static_ip.conf: Permission denied ]
^G Get Help     ^O Write Out    ^W Where Is     ^K Cut Text     ^C Cur Pos      ^Y Prev Page    M-\ First Line  ^B Back         ^A Home
^X Exit         ^R Read File    ^\ Replace      ^U Uncut Text   ^_ Go To Line   ^V Next Page    M-/ Last Line   ^F Forward      ^E End

ccrisan, 2015-01-23:
How did you login into your pi?

snyder250, 2015-01-23:
through Putty with the admin and password that i set up in motionpi

ccrisan, 2015-01-23:
I assume you use root as username. Admin isn't in fact supposed to be used with ssh.

snyder250, 2015-01-23:
I tried root as the login and "fe7f9ccf" for the password. I have to use the login and password I set up in the interface of notionpi ???

snyder250, 2015-01-23:
I don't want to take up to much of your time it is working now. I will just leave it up to DHCP. Thank you so much!!!

ccrisan, 2015-01-23:
Use "root" as username and your admin password set in UI prefs. For more details read [this](https://github.com/ccrisan/motionPie#shell-access).

snyder250, 2015-01-23:
You're a genius. That worked. I disabled the camera LED in config.txt and got the static IP working. Thank you

closed 2015-02-15 by ccrisan

omarmonn00, 2016-02-27:
Thanks ccrisan! This helped me too.