ccrisan/motionpie #80: ssh lockout

SuperOok, 2015-01-18:
Hi there,

I don't find a way to ssh into the motionpie. When I first started the pi I did not note the serial number. And first thing I did was renaming the "cryptic" host name in my router. So now I have to ssh into the pi but do not know the serial number for the password and have no clue how to get it. I did not find it in the web gui now that the first time welcome screen is gone...

I set the password for the admin user but this does not work as the root ssh password in the newest version.

So did I lock myself really out?

Cheers,
Hauke

rpilover, 2015-01-18:
log in as root via ssh, not admin

ccrisan, 2015-01-18:
First of all, your s/n will show up as a banner when trying to ssh into your pi.

Your GUI admin password will become the ssh root password when changed, but only after a reboot.

SuperOok, 2015-01-19:
Ok, thanks. The problem was, I was not using putty but some iPad ssh client that did neither show the banner nor the welcome text. With putty it worked.

closed 2015-01-19 by SuperOok