ccrisan/motionpie #81: Pure IP-Cam settings and performance

SuperOok, 2015-01-19:
Hi there,
I try to setup the Raspi Model B with a PiCam NoIR as a pure IP streaming client. All processing of stream can be done by my synology surveillance station on the stream receiving side.

What are the best possible settings to get a smooth stream with every other post processing disabled?

I disabled motion detection, text overlay and stream resizing. I find this resulting conf-file

```
 cat /data/etc/thread-1.conf
# @working_schedule_type outside
# @enabled on
# @name Camera1
# @webcam_resolution 100
# @network_share_name
# @network_server
# @motion_detection off
# @network_username
# @storage_device custom-path
# @preserve_movies 0
# @preserve_pictures 0
# @network_password
# @id 1
# @working_schedule
# @webcam_server_resize off

lightswitch 0
snapshot_interval 0
noise_level 31
ffmpeg_output_movies off
stream_motion off
target_dir /data/output
text_double off
height 720
width 1280
stream_quality 100
threshold 18179
quality 85
post_capture 4
on_event_end /programs/motioneye/eventrelay.py stop %t
text_left
pre_capture 2
noise_tune on
picture_filename
locate_motion_style redbox
locate_motion_mode off
contrast 153
stream_maxrate 15
output_pictures off
hue 0
saturation 64
stream_localhost off
videodevice /dev/video0
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes off
movie_filename %Y-%m-%d/%H-%M-%S
event_gap 10
auto_brightness off
stream_port 8081
rotate 270
text_right
on_event_start /programs/motioneye/eventrelay.py start %t
brightness 153
ffmpeg_bps 44000
framerate 15
emulate_motion off
minimum_motion_frames 1
snapshot_filename
```
Increasing the resolution at about 15fps also linearly increases CPU load. However, while the CPU load is for example at about 80% at the 1280x720 resolution, the frames per second seem quite limited; more like 2-3 fps than the requested 15fps.

Why is it only using 80% but showing only low fps?
Any other performance eating setting I can disable to get some decent stream?

Cheers,
Ook

ccrisan, 2015-01-20:
Don't attempt to use motionPie as a pure IP camera. There's a considerable amount of overhead regardless of the configuration choices you make.

closed 2015-01-20 by ccrisan

ccrisan, 2015-07-05:
For the record, [Fast Network Camera](https://github.com/ccrisan/motionpie/wiki/Fast-Network-Camera) is now available.