ccrisan/motionpie #82: Default settings for File Storage

dkelley810, 2015-01-20:
Storage Device - Custom Path
Root Directory /

is this right? I changed it while trying setup a network share.
I am worried that i missed something. and it will not save files properly.

Thanks Dennis

fgarciasa, 2015-01-20:
hello
I have a Banana Pi as a server with two raspberry Motion Pi Pie.
This is my setup:
[image]

ccrisan, 2015-01-20:
The share name must not contain slashes. It's probably just "datos". The rest can go to root directory (i.e. /cam/comedor2)

dkelley810, 2015-01-20:
I have tried a combination of all settings. Should i have to reboot after making changes? also when I go to use the network share, my preview of the camera just show the camera with line thru it. like its not configured.

ccrisan, 2015-01-20:
No need to reboot. You probably misconfigured your network share settings.

dkelley810, 2015-01-20:
Not sure if you can see the image. I was able to get the preview image back up with these settings.
[image]

But the Disk Usage, is the supposed to reflect the network share disk space?

ccrisan, 2015-01-20:
Yes, the disk usage should normally reflect the share disk space. If you think that's not the case for you, please post the output of the "df" and the "mount" commands from your motionPie.

dkelley810, 2015-01-20:
on my Pi or on the network share?

dkelley810, 2015-01-20:
dennis@antec1:/media/10GiBHD/MotionPie$ df
Filesystem                                             1K-blocks      Used Available Use% Mounted on
rootfs                                                  76921888  54494080  18520328  75% /
udev                                                       10240         0     10240   0% /dev
tmpfs                                                     200120      4444    195676   3% /run
/dev/disk/by-uuid/3109e512-dc1e-4774-9038-59a08612644c  76921888  54494080  18520328  75% /
tmpfs                                                       5120         0      5120   0% /run/lock
tmpfs                                                     400220        72    400148   1% /run/shm
/dev/sdb1                                               76920864    253944  72759492   1% /media/disk3
/dev/sdc1                                              488385532 436527784  51857748  90% /media/SIMPLE
/dev/sdd1                                                6196704    143384   5738544   3% /media/DELL8
/dev/sde1                                                9849376    153176   9195876   2% /media/10GiBHD

dennis@antec1:/media/10GiBHD/MotionPie$ mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,relatime,size=10240k,nr_inodes=217574,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,noexec,relatime,size=200120k,mode=755)
/dev/disk/by-uuid/3109e512-dc1e-4774-9038-59a08612644c on / type ext4 (rw,relatime,errors=remount-ro,user_xattr,barrier=1,data=ordered)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /run/shm type tmpfs (rw,nosuid,nodev,noexec,relatime,size=400220k)
/dev/sdb1 on /media/disk3 type ext4 (rw,relatime,errors=remount ro,user_xattr,barrier=1,data=ordered)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
/dev/sdc1 on /media/SIMPLE type fuseblk (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other,blksize=4096)
/dev/sdd1 on /media/DELL8 type ext4 (rw,relatime,errors=remount-ro,user_xattr,barrier=1,data=ordered)
/dev/sde1 on /media/10GiBHD type ext3 (rw,relatime,errors=remount-ro,barrier=1,data=ordered)
rpc_pipefs on /var/lib/nfs/rpc_pipefs type rpc_pipefs (rw,relatime)

ccrisan, 2015-01-20:
On motionPie. This doesn't look like motionPie.

dkelley810, 2015-01-20:
root@192.168.1.106's password: 
[root@mp-9026ce7f ~]# df
Filesystem                Size      Used Available Use% Mounted on
/dev/root               116.2M     91.3M     16.6M  85% /
devtmpfs                187.8M         0    187.8M   0% /dev
/dev/mmcblk0p1           16.0M      6.7M      9.3M  42% /boot
/dev/mmcblk0p3           14.0G     44.3M     13.2G   0% /data
tmpfs                   187.8M         0    187.8M   0% /dev/shm
tmpfs                   187.8M    560.0K    187.3M   0% /tmp
/dev/mmcblk0p3           14.0G     44.3M     13.2G   0% /home/ftp/sdcard
/dev/mmcblk0p3           14.0G     44.3M     13.2G   0% /home/ftp/storage
//192.168.1.85/MotionPie
                          9.4G    149.6M      8.8G   2% /data/media/motioneye_192_168_1_85_motionpie_dennis

[root@mp-9026ce7f ~]# mount
rootfs on / type rootfs (rw)
/dev/root on / type ext4 (ro,relatime,norecovery)
devtmpfs on /dev type devtmpfs (rw,relatime,size=192288k,nr_inodes=48072,mode=755)
proc on /proc type proc (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p3 on /data type ext4 (rw,noatime,data=ordered)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p3 on /home/ftp/sdcard type ext4 (rw,noatime,data=ordered)
/dev/mmcblk0p3 on /home/ftp/storage type ext4 (rw,noatime,data=ordered)
//192.168.1.85/MotionPie on /data/media/motioneye_192_168_1_85_motionpie_dennis type cifs (rw,relatime,vers=1.0,cache=strict,username=dennis,domain=ANTEC1,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.1.85,unix,posixpaths,serverino,acl,rsize=1048576,wsize=65536,actimeo=1)

ccrisan, 2015-01-20:
Alright, let some media files pile up on your network share and see if the usage increases. I use the "statvfs" call to determine the disk usage and that might be different from what df uses.

dkelley810, 2015-01-21:
if i do not use a network share. what are the settings if i want to put it back to use the default settings?

ccrisan, 2015-01-21:
The default values for storage are "Custom Path" with root directory "/data/output".

closed 2015-01-21 by ccrisan

dkelley810, 2015-01-29:
I am trying to get this to work with my network share. when I attempt to make the same setting that I mention worked above. Are no longer working. I get no camera active. Now I made the the changes, and rebooted. All I get is a spinning circle of dots. is it going to eventually reboot or should I power it down. last time that i pulled the power, it never came back on. and I needed to redo the sdcard.

ccrisan, 2015-01-29:
What are the exact settings that you use when configuring the network share, @dkelley810? I would like to see all of them, except for the password.

dkelley810, 2015-01-30:
Are you looking for information from my other computer on my network that I want to store the images and videos to? Not the on the Pi?

dkelley810, 2015-01-30:
ok I got it to work.
on my remote system I have a hard drive mounted at /media/10GiBHD
[image]

[root@mp-9026ce7f ~]# df
Filesystem                Size      Used Available Use% Mounted on
/dev/root               116.2M     91.2M     16.6M  85% /
devtmpfs                187.8M         0    187.8M   0% /dev
/dev/mmcblk0p1           16.0M      6.7M      9.3M  42% /boot
/dev/mmcblk0p3           14.0G     56.8M     13.2G   0% /data
tmpfs                   187.8M         0    187.8M   0% /dev/shm
tmpfs                   187.8M    896.0K    187.0M   0% /tmp
df: /home/ftp/sdcard\040(deleted): No such file or directory
/dev/mmcblk0p3           14.0G     56.8M     13.2G   0% /home/ftp/storage
//192.168.1.85/10GiBHD
                          9.4G      3.5G      5.4G  40% /data/media/motioneye_192_168_1_85_10gibhd_dennis

[root@mp-9026ce7f ~]# mount
rootfs on / type rootfs (rw)
/dev/root on / type ext4 (ro,relatime,norecovery)
devtmpfs on /dev type devtmpfs (rw,relatime,size=192288k,nr_inodes=48072,mode=755)
proc on /proc type proc (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p3 on /data type ext4 (rw,noatime,data=ordered)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p3 on /home/ftp/sdcard\040(deleted) type ext4 (rw,noatime,data=ordered)
/dev/mmcblk0p3 on /home/ftp/storage type ext4 (rw,noatime,data=ordered)
//192.168.1.85/10GiBHD on /data/media/motioneye_192_168_1_85_10gibhd_dennis type cifs (rw,relatime,vers=1.0,cache=strict,username=dennis,domain=ANTEC1,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.1.85,unix,posixpaths,serverino,acl,rsize=1048576,wsize=65536,actimeo=1)

MaxTPS, 2015-02-18:
Hi there, 

I'm testing this excellent MotionPie (and btw, thank you, it works like a charm).
However, I have a issue for the network storage. See my configuration below. 
I'm trying to store files on my NAS Synology. 
Users & rights are rightly configured. I can access my share through Windows network discovery but impossible to have an image through the PiCam when I select Network Share in Motion Pie. 
Any clue ? 
[image]

ccrisan, 2015-02-18:
What does the path in the address bar look like when you access the share in Windows Explorer?

MaxTPS, 2015-02-18:
\\<name_of_the_nas>\RaspCam1 

i've already try to put the full path in the root directory with no luck...

ccrisan, 2015-02-18:
That's not the full path. It can't be. You need a server (name or IP) and a share name. "RaspCam1" looks like the name of a network server, although the name is a bit misleading (giving the false idea that it's the raspberry pi itself). Following that, you need a share name, which you must have previously configured.

To help you and others like you having trouble configuring a network share, I have put together a section dedicated to [network shares](https://github.com/ccrisan/motionPie/wiki/Troubleshooting#network-shares) in the troubleshooting wiki page.

MaxTPS, 2015-02-18:
"RaspCam1" is the name of the shared folder which is now renamed in "recording" so there's no misleading anymore. 
Here's the full path with IP adress instead of the hostname. 
I've followed your troubleshooting page but i can't see where i screw up... 
[image]

[image]

ccrisan, 2015-02-18:
Alright this looks good to me. What are the symptoms now? Can I see the /var/log/motioneye.log and /var/log/motion.log files?

MaxTPS, 2015-02-18:
i'm completely newbie for this...
how to ? 
can i connect through ssh (putty) to get those files ?

ccrisan, 2015-02-18:
You can use [filezilla](https://filezilla-project.org/) to connect to the SFTP server. There, you can browse all the files on your system.

MaxTPS, 2015-02-18:
damnit i must be cursed ! 
cannot connect with filezilla.... even if i change the admin password
[image]

ccrisan, 2015-02-18:
What password do you use? Have you set an admin password in the UI or have you left it empty? If you changed it, you need to reboot before it takes effect.

MaxTPS, 2015-02-18:
got it... the reboot did the trick. thx. 
but no log out there.... 
[image]

ccrisan, 2015-02-18:
You're in /home/ftp/storage/... Logs are in /var/log/.

MaxTPS, 2015-02-18:
ashamed is how i feel at the moment.... 
well, here's the logs... finally! 

motion.log
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] motion_startup: Motion trunkREV561 Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (EMG)

motioneye.log
2015-02-18 07:49:46: INFO: main config file /data/etc/motion.conf does not exist, using default values
2015-02-18 07:49:46: INFO: cleanup started
2015-02-18 07:49:46: INFO: wsswitch started
2015-02-18 07:49:46: INFO: thumbnailer started
2015-02-18 07:49:47: INFO: server started
2015-02-18 07:49:49: INFO: adding new camera with id 1...
2015-02-18 07:49:49: WARNING: control hue not found for device /dev/video0
2015-02-18 07:49:49: INFO: main config file /data/etc/motion.conf does not exist, using default values
2015-02-18 07:49:51: WARNING: control hue not found for device /dev/video0
2015-02-18 08:22:44: WARNING: control hue not found for device /dev/video0
2015-02-18 08:22:44: WARNING: control hue not found for device /dev/video0
2015-02-18 08:22:54: WARNING: control hue not found for device /dev/video0
2015-02-18 08:22:54: WARNING: control hue not found for device /dev/video0
2015-02-18 08:23:09: ERROR: authentication failed for user admin
2015-02-18 08:23:10: ERROR: authentication failed for user admin
2015-02-18 08:23:17: WARNING: control hue not found for device /dev/video0
2015-02-18 08:23:17: WARNING: control hue not found for device /dev/video0
2015-02-18 08:23:32: WARNING: control hue not found for device /dev/video0
2015-02-18 08:23:39: WARNING: control hue not found for device /dev/video0
2015-02-18 08:24:08: WARNING: control hue not found for device /dev/video0
2015-02-18 08:24:14: WARNING: control hue not found for device /dev/video0
2015-02-18 08:24:48: WARNING: control hue not found for device /dev/video0
2015-02-18 08:24:53: WARNING: control hue not found for device /dev/video0
2015-02-18 08:27:26: WARNING: control hue not found for device /dev/video0
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-02-18 08:27:31: ERROR: failed to mount smb share "//192.168.1.200/RaspCam1" at "/data/media/motioneye_192_168_1_200_raspcam1_admin"
2015-02-18 08:27:31: WARNING: control hue not found for device /dev/video0
2015-02-18 08:27:42: WARNING: control hue not found for device /dev/video0
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-02-18 08:27:42: ERROR: failed to mount smb share "//192.168.1.200/RaspCam1" at "/data/media/motioneye_192_168_1_200_raspcam1_cloudtps_admin"
2015-02-18 08:27:42: WARNING: control hue not found for device /dev/video0
2015-02-18 08:27:51: WARNING: control hue not found for device /dev/video0
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-02-18 08:27:51: ERROR: failed to mount smb share "//192.168.1.200/RaspCam1" at "/data/media/motioneye_192_168_1_200_raspcam1_cloudtps_admin"
2015-02-18 08:27:52: WARNING: control hue not found for device /dev/video0
2015-02-18 08:28:01: WARNING: control hue not found for device /dev/video0
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-02-18 08:28:01: ERROR: failed to mount smb share "//192.168.1.200/RaspCam1" at "/data/media/motioneye_192_168_1_200_raspcam1_cloudtps_admin"
2015-02-18 08:28:01: WARNING: control hue not found for device /dev/video0
2015-02-18 08:28:01: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_1_200_raspcam1_cloudtps_admin/volume2/RaspCam1'
2015-02-18 08:28:17: WARNING: control hue not found for device /dev/video0
2015-02-18 08:28:17: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_1_200_raspcam1_cloudtps_admin/volume2/RaspCam1'
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-02-18 08:28:17: ERROR: failed to mount smb share "//192.168.1.200/RaspCam1" at "/data/media/motioneye_192_168_1_200_raspcam1_cloudtps_admin"
2015-02-18 08:28:17: WARNING: control hue not found for device /dev/video0
2015-02-18 08:28:17: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_1_200_raspcam1_cloudtps_admin/volume2'
2015-02-18 08:28:49: WARNING: control hue not found for device /dev/video0
2015-02-18 08:28:49: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_1_200_raspcam1_cloudtps_admin/volume2'
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-02-18 08:28:49: ERROR: failed to mount smb share "//192.168.1.200/RaspCam1" at "/data/media/motioneye_192_168_1_200_raspcam1_cloudtps_admin"
2015-02-18 08:28:49: WARNING: control hue not found for device /dev/video0
2015-02-18 08:28:49: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_1_200_raspcam1_cloudtps_admin/volume2'
[... 377 lines cut ...]
2015-02-18 13:38:10: INFO: interrupt signal received, shutting down...
2015-02-18 13:38:10: INFO: server stopped
2015-02-18 13:38:12: INFO: interrupt signal received, shutting down...
[... cut ...]
2015-02-18 13:43:30: ERROR: failed to mount smb share "//192.168.1.200/recording" at "/data/media/motioneye_192_168_1_200_recording_admin"
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-02-18 13:48:30: ERROR: failed to mount smb share "//192.168.1.200/recording" at "/data/media/motioneye_192_168_1_200_recording_admin"
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
2015-02-18 13:53:30: ERROR: failed to mount smb share "//192.168.1.200/recording" at "/data/media/motioneye_192_168_1_200_recording_admin"

ccrisan, 2015-02-18:
Well, there are a lot of "permission denied" messages that suggest... well, that credentials are not accepted. Check that  the username and password are correct (retype them), making sure they don't have spaces around them and that the case is the right one.

MaxTPS, 2015-02-18:
yep, this is obvious...
i tried with a new user called "administrator" and a stupid password such as 1122334455
so i cannot be wrong with the case
but the symptoms remains the same...

ccrisan, 2015-02-18:
Can you ssh into your motionPie (using putty), and run the following command:

    mkdir /data/something && mount.cifs //192.168.1.200/recording /data/something -o username=administrator,password=1122334455

What does the command say? Does it work right away or gives the same permission denied error?

MaxTPS, 2015-02-18:
unfortunately... permission denied... 
[image]

ccrisan, 2015-02-18:
I have no further ideas. I'm sorry. When you try to access this share from a Windows machine, are you prompted for username and password? If you enter these very same credentials does it work?

MaxTPS, 2015-02-18:
yes, i've once again verified the credentials and try to map the share with 3 differents machines. (XP, Win7 and Win8) .... All three are working and yes i'm prompted for a username and password...

ccrisan, 2015-02-18:
Do you have a domain name for your Synology? Try to run the same command but with an additional `domain` argument:

    mount.cifs //192.168.1.200/recording /data/something -o domain=YourDomain,username=administrator,password=1122334455

And just out of curiosity, try it again with guest:

    mount.cifs //192.168.1.200/recording /data/something -o guest

MaxTPS, 2015-02-18:
yes i do have a domain name for the Syno. but the credentials i use are local (the default admin of the syno)

i've tried your command with the domain name just to be sure but same issue. 

with the guest account, i have this error message... 

[image]

MaxTPS, 2015-02-19:
Hi Ccrisan! 
After a good night sleeping, i've decided to try with another Synology nas with a fresh basic installation/configuration. That worked immediately! 
So i tried again with my main NAS and it worked instantly. I don't understand where i've make a mistake but the thing is that now, it works. 

I guess I needed some sleep to figure it out :-) 

Now i have two questions which are: 
1° would it be possible to make the red led working only during motion detection ? 
2° what is the configuration point i've got to modify to get a smooth video ? 
I understand that during the Stream, it's a bit jerky but on the .avi it's the same and i'd like to get a good video quality. would that be possible ? (i don't care if it takes much storage, i have plenty of it)

ccrisan, 2015-02-19:
1) no (at least I don't know how to do that)
2) you won't get a smooth video with an old raspberry pi; your best bet is to decrease the resolution as much as possible

MaxTPS, 2015-02-19:
tought so.... gonna buy some raspberry 2 
thank you anyway for your time spent! ;-)

ccrisan, 2015-02-19:
You're welcomed. I'm sorry for the possibly unsatisfying answers but these are the real limitations.

NathTheDude, 2015-06-18:
Hi,

I appreciate this is an older post but thought I would throw something in there. At first I was unable to get the sharing working. After going through this post I got a workaround which is fine, for some reason the username and password I was using did not work due to the special characters I had in the password. When I changed this to a simple password it worked straight away.

If you wanted to get the LED to flash when motion is detected you may be able to add a script to run when motion is detected turn on the led.....just a thought.

ccrisan, 2015-06-19:
@NathTheDude I'd like to know what special characters you used and did not work; maybe I can fix that.

NathTheDude, 2015-06-19:
Thanks for the reply Calin, honestly it's not a problem :)

I used these special characters:
@*$