ccrisan/motionpie #83: Login Loop

bonjoh, 2015-01-21:
I'm trying to add a camera. As I'm in the add camera dialog, and enter the URL of my camera and tab/click on the username field for the camera, the system login modal pops up, and won't accept any username/passwords.

I'm stuck here, unable to add a network camera, because the popup keeps looping and preventing me from adding the camera, saying my login is invalid.  I've tried this without changing the password from blank, a simple password, even the camera's authentication, and can't get a successful login at that modal.  I've tried a few different images created with writeimage.sh on OSX 10.10.1

This happens when I enter any url with 'http://'. If I just use a URL sans-http, it doesn't do the login loop but it also doesn't proceed past the Add Camera dialog, since the URL is invalid.

ccrisan, 2015-01-21:
What version do you currently run? What is the URL of the camera you're trying to add? There might be a problem with special characters inside this URL that might mess up the signature algorithm.

bonjoh, 2015-01-21:
Current Version is 20150112, and I've tried a few different camera urls from a few different cameras, including 'http://192.168.1.140/axis-cgi/jpg/image.cgi'

jarno1977, 2015-01-27:
I have exactly the same problem with version 20150112
The url I use is http://192.168.2.141/videostream.cgi?user=admin&pwd=admin

If its help I could provide more details/logs. please let me know!

ccrisan, 2015-02-08:
There was a bug indeed when the URI of the netcam had query arguments. It's fixed now and the fix will be included in the next motionPie version. Thanks for reporting the issue.

closed 2015-02-08 by ccrisan