ccrisan/motionpie #85: traceback

piecam, 2015-01-23:
HI,

What might be the reason of this? it's pi b+, with pi cam of resoultion setting 1600x1200.
tks
2015-01-21 07:19:04: WARNING: Connect error on fd 8: ECONNREFUSED
2015-01-21 07:19:04: WARNING: Connect error on fd 8: ECONNREFUSED
2015-01-21 07:19:04: WARNING: control hue not found for device /dev/video0
2015-01-21 07:19:11: ERROR: could not terminate the motion process
Traceback (most recent call last):
  File "media/data/pi/motionPie/output/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1115, in _stack_context_handle_exception
  File "media/data/pi/motionPie/output/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1298, in wrapper
  File "/programs/motioneye/src/handlers.py", line 183, in post
    self.set_config(camera_id)
  File "/programs/motioneye/src/handlers.py", line 119, in wrapper
    return func(self, *args, **kwargs)
  File "/programs/motioneye/src/handlers.py", line 378, in set_config
    set_camera_config(int(key), cfg, check_finished)
  File "/programs/motioneye/src/handlers.py", line 267, in set_camera_config
    on_finish(None, True) # (no error, motion needs restart)
  File "/programs/motioneye/src/handlers.py", line 368, in check_finished
    finish()
  File "/programs/motioneye/src/handlers.py", line 337, in finish
    motionctl.stop()
  File "/programs/motioneye/src/motionctl.py", line 159, in stop
    raise Exception('could not terminate the motion process')
Exception: could not terminate the motion process
2015-01-21 07:19:11: ERROR: 500 POST /config/0/set/?_=1421842744312&_username=admin&_signature=b96ae1086e9f860ef282b368465e0950cf4ba892 (192.168.100.113) 7163.96ms
2015-01-21 07:19:12: WARNING: Connect error on fd 8: ECONNREFUSED
2015-01-21 07:19:12: WARNING: Connect error on fd 8: ECONNREFUSED

ccrisan, 2015-01-23:
At such high resolutions it's probably the motion daemon that crashes or becomes unresponsive. I strongly suggest decreasing the resolution.

closed 2015-01-23 by ccrisan

piecam, 2015-01-23:
HI ,
I do have anothe same identical setting for a NoIR cam and it looks like that unit is running fine.
I am just curious. May be this traceback is random?

ccrisan, 2015-01-23:
As I said, It looks like the motion daemon has crashed. Most likely motionEye would have tried to respawn it within a few seconds.