ccrisan/motionpie #87: Possible added feature

dkelley810, 2015-01-25:
is there a way to implement a button that will start a video. I was watching the live feed and a deer came into view. it would be nice to be able to start recording without motion being detected. I am working on my settings to detect motion. but what i was looking at was too dark, to show changes.

maybe place it next to the Full-screen window, pictures, movies, and configure buttons.

[image]

ccrisan, 2015-01-25:
I have in plan to implement such a features but it requires some serious workarounds when it comes to motion daemon and motion detection. I'm not promising anything

Godzookee, 2015-03-18:
I don't know if it is the same thing I was looking for a way to make it run all the time and just overwrite the files after X period or when it runs out of space.

Do you think this is possible?

Keep up the great work love the project.