ccrisan/motionpie #89: unable to open Samba share on my network

dkelley810, 2015-01-26:
I used the command below to create the sdcard. I enabled the internal Samba server this time. (-a writable). I am unable to get to it thru my network. I was able to change the workgroup name in smb.conf
is there a way to change it without writing the sdcard?

sudo ./writeimage.sh -d /dev/sdb -i ~/Downloads/motionpi/motionPie-20150112.img -a writable -l -n 2WIRE214:9613910036 -o modest

ccrisan, 2015-01-26:
What do you mean you're unable to get it thru your network? How exactly are you trying to access the samba server on your PI? From what OS, from what utility, what share and what credentials?

dkelley810, 2015-01-26:
I changed my smb.conf file:
[global]
   workgroup = KELLEY
   server string = motionPie
   security = user
   map to guest = bad user
   encrypt passwords = yes
   private dir = /var/lib/samba/private

[sdcard]
   comment = SD Card Output Directory
   path = /data/output
   public = no
   writable = yes

[storage]
   comment = Attached Storage Devices
   path = /data/media
   public = no
   writable = yes

I previously had public to yes. once I changed it to no, its working.

But what is wierd id when i click on the movies,pictures, and full screen window. It Prompts me for my login information. No matter what i enter for a username and password, does not allow me to login.

[image]

is it possible that I am running the SAMBA server on the Pi and on another computer on my network? How do I disable it on the Pi, if that is the case, from the command line?

Thank You again for all your work!

ccrisan, 2015-01-26:
1. You haven't answered my questions. Here they are again: What do you mean you're unable to get it thru your network? How exactly are you trying to access the samba server on your PI? From what OS, from what utility, what share and what credentials?

2. What is the connection between the samba problem and the problem you're describing regarding the login prompt?

3. Did you change the admin username?

dkelley810, 2015-01-26:
never mind. I figured out what I did. I had to click on 'open with' and the pick files to open with the right program.

I am using Linux Mint with NEMO 2.2.4 - once I opened it with the proper application and its working.

dkelley810, 2015-01-27:
so when I click on the movies,pictures, and full screen window. It Prompts me for my login information. No matter what i enter for a username and password, does not allow me to login.

<See Picture above>
if you think that it is a different issue from the Samba issue that i thought that i had. I will make it another issue.

ccrisan, 2015-01-27:
It certainly is a different issue, but leave it here for now. Could you send me your /data/etc/motion.conf and /data/etc/thread-1.conf?

You can remove passwords and other sensible information.

sh2515, 2015-01-27:
Have you tried admin and the serial number - same as you would use for ssh? I had similar issue and that resolved it for me

dkelley810, 2015-01-27:
/data/etc/motion.conf

# @enabled on
# @show_advanced on
# @normal_password **********
# @admin_username admin
# @admin_password *******
# @normal_username dennis

thread thread-1.conf
webcontrol_port 7999

/data/etc/thread-1.conf

# @working_schedule_type outside
# @enabled on
# @name Camera1
# @webcam_resolution 100
# @network_share_name media
# @network_server 192.168.1.85
# @motion_detection on
# @network_username dennis
# @storage_device custom-path
# @preserve_movies 0
# @preserve_pictures 0
# @network_password ********
# @id 1
# @working_schedule
# @webcam_server_resize off

lightswitch 0
snapshot_interval 0
noise_level 31
ffmpeg_output_movies on
stream_motion on
target_dir /data/output
text_double on
height 800
width 1280
stream_quality 100
threshold 0
quality 85
post_capture 5
on_event_end /programs/motioneye/eventrelay.py stop %t
text_left Camera1
pre_capture 5
noise_tune on
picture_filename %Y-%m-%d/%H-%M-%S
locate_motion_style redbox
locate_motion_mode on
contrast 168
stream_maxrate 6
output_pictures on
hue 0
saturation 0
stream_localhost off
videodevice /dev/video0
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
event_gap 10
auto_brightness off
stream_port 8081
rotate 90
text_right %Y-%m-%d\n%T
on_event_start /programs/motioneye/eventrelay.py start %t; /programs/motioneye/sendmail.py "smtp.gmail.com" "587" "dman81032@gmail.com" "ILoveBeverly09" "True" "dman81032@gmail.com" "motion_start" "%t" "%Y-%m-%dT%H:%M:%S"
brightness 166
ffmpeg_bps 2935466
framerate 10
emulate_motion off
minimum_motion_frames 1
snapshot_filename

dkelley810, 2015-01-27:
sh2515 - No change. Thanks for the idea.

ccrisan, 2015-01-27:
@dkelley810 I hope that's not your real gmail password. I'll try to reproduce the problem myself. Also, could you please tell me if your admin password has special characters in it? Spaces? Only numbers?

dkelley810, 2015-01-28:
my admin password does not contain spaces or special characters.
I missed the password in the thread-1.conf but i was able to change it.

dkelley810, 2015-01-28:
well today its working. I was able to ssh into the pi, and in the /data/output directory I found a folder named /RUPMRV~F and I don't know if this could be be the issue. But I was able to remove that directory and it seems to be working. now when I click on the icon, it says no media files if there are none. and does list the pictures and videos if there are.

ccrisan, 2015-01-29:
The RUPMRV~F directory looks like something samba would create when unsupported characters are involved. motionEye should behave nicer with those files - I'll try to reproduce the problem locally and come up with a fix for those cases.

ccrisan, 2015-02-08:
There was an issue when encountering special characters such as "~" in paths. It is now fixed and the fix will be included in the next release.

closed 2015-02-08 by ccrisan