ccrisan/motionpie #90: recommended best settings

piecam, 2015-01-28:
HI,
Can we have a post dedicated for people posting their best successful setting with MP? Are you considering setting up a wiki site for these so that people can learn from each other?

For example, I have tried various things for myself.

1. Pi B+, Pi IR Cam
1.1 With 1600x1200, it works most of day. It will give me a complete darkness the next morning or very bright "white". CPU is high around 90%(motion is 70-80%, motioneye the rest). I have to change resolution to some thing lower and set back to this to get correct color.
1.2 With 1280x1024, similiar as above, except cpu is a bit lower.
1.3 with 1280x720, things look good next day, color is right.
1.4 My drive way is 50feet deep, I need minimum 1280x1024 to clearly identify car plate number. 1280x720 doesn't do that.
1.5 frame rate is 2/sec
1.6 motion detection is 5% for all setting, 2% gives more more pic.
1.7 use motion triggered still image
1.8 rest are defaults

2. Pi B+, Pi NoIR Cam
2.1 similiar like Pi IR cam, except that picture the next day will either be "white" or "greeny", with snow or sun effect.
2.2 with 1280x720, the color looks right next day
2.3 Without 1280x1024 or above, can't identiy plate number
2.4 frame rate is 2/sec
2.5 motion detection is 5% for all setting, 2% gives more more pic.
2.6 use motion triggered still image
2.7 rest are defaults

3. Pi B+, Logitech Pro 9000
3.1 resolutio 1600x1200, CPU is fine. Not sure this is true 1600x1200 comparing to Pi Cam.
3.2  image color the next day is correct.
3.3 frame rate is 2/sec
3.4 motion detection is 10% for all setting, trying to avoid backyard tree movemoent.
3.5 use motion triggered still image
3.6 rest are defaults

4. Pi B+, logitec c270
tba

ccrisan, 2015-01-28:
I guess I could organize things around using wiki pages.

robertybob, 2015-01-28:
+1 , sounds like a great idea!

bonjoh, 2015-01-28:
+1 I'm totally on board, too, and would love to help out when needed!

piecam, 2015-02-05:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Here'are some of screen captures.
With 1600x1200 for Pi IR Cam, image got into a state of too bright under sun or snow conditions. with 1280x720, it looks fine.
With 1600x1200 for Pi NoIR cam, image got greeny at bright sun or snow. with 1280x720, it looks ok.
with 1600x1200 for Logitech Pro 9000 old USB, it looks fine.
The other USB is logitech c270, which works fine with 1280x960 at front door.

Anyone has any ideas why USB cam (9000 or 270) works better (perceived) than Pi Cam (IR or NoIR) at higher resolution? Two PiCam Units outdoor and two indoors. Outside temperature is -14C with wind chill is about -28C.

I have difficuty in getting high resulution for this Pi Cam to get same quality as at least USB pro 9000 cam.

dkelley810, 2015-02-09:
I am using a Raspberry Pi B with a NoIR Pi Cam attached.
I have two sets of settings that I use. one for Day and one for Night.
[image]
[image]

Night -
[image]

ccrisan, 2015-02-09:
Could you please take this to the [wiki](https://github.com/ccrisan/motionPie/wiki/Recommended-Settings)? It's not an issue per se and I believe it belongs to a wiki page. Thanks.

dkelley810, 2015-02-15:
This is the area to post the settings that work for your setup. Please post settings and screenshots. Please list version of Raspberry Pi you have, what type of camera/cameras you have connected. One thing I would like to know is, people with multi camera setups, how is it all physically connected? This may not be the ideal place for this information, but I feel it is helpful to myself and to future users of MotionPie.

piecam, 2015-02-16:
will try to post soon. tks

lastfreenick, 2015-02-18:
I run motionpi with the motion mmal binary and stills mode enabled in motion's config file. Good thing is no brightness hanging issues at all resolutions and very good low light performance. Only real downside is a super slow framerate like 0.3 - 0.5 frames per second.

piecam, 2015-02-18:
which file is it to change to mmal and stills mode? I have brightness issuse with resolution over 1280x720 for Pi IR Cam or NoIR cam, not for usb cam like logitech 9000 or c270. tks

piecam, 2015-02-18:
please help details the steps to convert to Mmal version. thanks in advance.

piecam, 2015-02-18:
woud converting to mmal affect Pis with Pi Cam+USB cam? I have a 2 with both Pi Cam and Usb cam.

thanks in advance

lastfreenick, 2015-02-18:
Ok now here's the noobs guide to enabling motion_mmal's stills mode

First one should ssh into the motionpi machine

ssh root@your.pi.ip

once connected add the line "mmalcam_use_still on" to the motion configuration file, the conf file is located at /data/etc/ and is called thread-1.conf. You can edit the file with the following command

nano /data/etc/thread-1.conf 

the add

mmalcam_use_still on

at the bottom of the file, save and exit.

Then go to 

/usr/bin/

there you can find two files called motion and motion-mmal, those are the binary files. motion-mmal will only be there if you are on the latest (20150215) relase of motionpi. What needs to be done now is motion needs to be renamed motion.old (or some other name that is NOT motion) and motion-mmal needs to be renamed motion. To be able to rename the files one first needs to remount the filesystem as read/write which is done by

mount -o remount,rw /dev/root /

onve this is done rename the files

mv motion motion.old
mv motion-mmal motion

one last step is to reboot by typing

reboot

after the reboot, your picam should be using motion-mmal binary in stills mode which gives you no more coulor/brightness mess up, very nice low light performance and a very slow framerate of about 0.2 - 0.5 frames per second because the pi's camera sensor is recalibrated ( I guess so) for every new frame, hence the improved performance.

To answer piecam's last question: I have absolutely no idea, try it for yourself ;)

lastfreenick, 2015-02-18:
From what I've read in other issues comments calin is probalby going to add a gui feature to switch between motion and motion_mmal binaries, also someone seems to have found a way around the cam brightness/colour issues at high resolutions and these might soon be implemented by our ever so industrious host. In theory, this would make this whole messing around with motion_mmal unnecessary except for one 'feature' of stills mode which from my experimenting is the low light performance. When stills mode is enabled, I feel the cam captures much better pictures in low light/darkness probably due to longer exposure times in stills mode. Those longer exposure times mean a very slow framerate, but some people might prefer brightness over framerate.

piecam, 2015-02-18:
Much appreciated! will test it out! many thanks!

ZeroT, 2015-02-19:
Has anyone run this on both the Pi and Pi2?  Is the Pi2 performance a lot better as the Pi is very disappointing framerate wise.

ccrisan, 2015-02-19:
The performance is not necessarily *a lot better*, but there is a sensible improvement. I'd say it works about twice as fast as it does on an old PI.

piecam, 2015-02-19:
I have two Pi B+ with Pi IR and NoIR cam, with mmal enabled. However, I just put it for a few hours. Its still two bright for Pi IR cam and greeny for Pi NoIR cam with resolution above 1280x720. Any other obvious setting I have to put in to get good quality hi res pic without these "bright/greeny" issue.?
thanks

These are two captures of Pi IR cam with 1280x720 and 1600x1200 in bright sunshine/snow. It looks 1280x720 is better. Not sure whether this is soly by high res pick up more lights?!

[image]
[image]

This is PI NoIR cam, with 1280x720 and 1600x1200. same sunshine/snow. 1280x720 above are all greeny.
[image]
[image]

piecam, 2015-02-20:
HI lastfreenick 

Can you share your other settings? Are you on 1600x1200? Are you with IR cam or NoIR cam. It still looks too "white/bright" on IR cam, and "greeny" on NoIR at high resolution. I haven't changed saturation/bright/contrat (50%) default.

thanks

lastfreenick, 2015-02-21:
I've completely diabled contrast, brightness and saturation in the motion conf file (data/etc/thread-1.conf) by setting the values in the conf file to zero. If you use the sliders in motionpi 1 is the lowest setting possible, setting it to 0 disables these settings alltogether. Another thing I've added to the conf file is the line mmalcam_name vc.ril.camera which I took from the mmal example conf, not sure if or what it does though

lastfreenick, 2015-02-21:
Also you might want to check /data/log/motion.log line three should read motion mmaltest started when correctly running mmal motion binary. Those greenish and ovexposed pics look like either motion mmal or stills mode isnt running correctly

piecam, 2015-02-21:
It looks like that those already been set to "0", probably by mmal???Even though my GUI setting is still 50%.

[root@mp-e8e50abb ~]# cat /data/etc/thread-1.conf | grep 0
# @webcam_resolution 100
# @network_server 192.168.100.10
# @preserve_movies 0
# @preserve_pictures 30
lightswitch 0
snapshot_interval 0
target_dir /data/media/motioneye_192_168_100_10_recording_cam
height 1200
width 1600
threshold 96000
contrast 0
hue 0
saturation 0
videodevice /dev/video0
ffmpeg_variable_bitrate 0
event_gap 10
stream_port 8081
rotate 0
brightness 0
ffmpeg_bps 44000
[root@mp-e8e50abb ~]#

piecam, 2015-02-21:
I have this mmal enabled for PiB+ with Pi NoIR can+USB quickpro 9000. It looks like that on both configus, the setting are set to "0", even though the GUI is 50%.
[root@mp-634ef108 ~]# cat /data/etc/thread-1.conf | grep 0
# @webcam_resolution 100
# @network_server 192.168.100.10
# @preserve_movies 0
# @preserve_pictures 30
lightswitch 0
snapshot_interval 0
target_dir /data/media/motioneye_192_168_100_10_recording_cam
height 1200
width 1600
contrast 0
hue 0
saturation 0
videodevice /dev/video0
ffmpeg_variable_bitrate 0
event_gap 10
stream_port 8081
rotate 0
brightness 0
ffmpeg_bps 44000
[root@mp-634ef108 ~]# cat /data/etc/thread-2.conf | grep 0
# @webcam_resolution 100
# @network_server 192.168.100.10
# @preserve_movies 0
# @preserve_pictures 30
lightswitch 0
snapshot_interval 0
target_dir /data/media/motioneye_192_168_100_10_recording_cam
height 1200
width 1600
threshold 96000
contrast 0
hue 0
saturation 0
ffmpeg_variable_bitrate 0
event_gap 10
stream_port 8082
rotate 0
brightness 0
ffmpeg_bps 44000

piecam, 2015-02-21:
Here's full two configs for NoIR am and usb cam. fyi
[root@mp-634ef108 ~]# cat /data/etc/thread-2.conf         
# @working_schedule_type outside
# @enabled on
# @name Camera2
# @webcam_resolution 100
# @network_share_name Recording
# @network_server 192.168.100.10
# @motion_detection on
# @network_username cam
# @storage_device network-share
# @preserve_movies 0
# @preserve_pictures 30
# @network_password ccie8577
# @id 2
# @working_schedule 
# @webcam_server_resize off

lightswitch 0
snapshot_interval 0
noise_level 31
ffmpeg_output_movies off
stream_motion off
target_dir /data/media/motioneye_192_168_100_10_recording_cam
text_double on
height 1200
width 1600
stream_quality 85
threshold 96000
quality 85
post_capture 4
on_event_end /programs/motioneye/eventrelay.py stop %t
text_left Rear Window
pre_capture 2
noise_tune on
picture_filename %Y-%m-%d/Rearwindow/%H-%M-%S
locate_motion_style redbox
locate_motion_mode on
contrast 0
stream_maxrate 5
output_pictures on
hue 0
saturation 0
stream_localhost off
videodevice /dev/video1
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
event_gap 10
auto_brightness off
stream_port 8082
rotate 0
text_right %Y-%m-%d\n%T
on_event_start /programs/motioneye/eventrelay.py start %t
brightness 0
ffmpeg_bps 44000
framerate 5
emulate_motion off
minimum_motion_frames 1
snapshot_filename 
[root@mp-634ef108 ~]# cat /data/etc/thread-1     
cat: can't open '/data/etc/thread-1': No such file or directory
[root@mp-634ef108 ~]# cat /data/etc/thread-1.conf 
# @working_schedule_type outside
# @enabled on
# @name Camera1
# @webcam_resolution 100
# @network_share_name Recording
# @network_server 192.168.100.10
# @motion_detection on
# @network_username cam
# @storage_device network-share
# @preserve_movies 0
# @preserve_pictures 30
# @network_password ccie8577
# @id 1
# @working_schedule 
# @webcam_server_resize off

lightswitch 0
snapshot_interval 0
noise_level 31
ffmpeg_output_movies off
stream_motion off
target_dir /data/media/motioneye_192_168_100_10_recording_cam
text_double on
height 1200
width 1600
stream_quality 85
threshold 97917
quality 85
post_capture 4
on_event_end /programs/motioneye/eventrelay.py stop %t
text_left Rear_Door
pre_capture 2
noise_tune on
picture_filename %Y-%m-%d/Reardoor/%H-%M-%S
locate_motion_style redbox
locate_motion_mode on
contrast 0
stream_maxrate 5
output_pictures on
hue 0
saturation 0
stream_localhost off
videodevice /dev/video0
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
event_gap 10
auto_brightness off
stream_port 8081
rotate 0
text_right %Y-%m-%d\n%T
on_event_start /programs/motioneye/eventrelay.py start %t
brightness 0
ffmpeg_bps 44000
framerate 2
emulate_motion off
minimum_motion_frames 1
snapshot_filename 
mmalcam_use_still on
[root@mp-634ef108 ~]#

piecam, 2015-02-21:
oops, got my password exposed!!! will change now!! :-)

piecam, 2015-02-21:
These are captures of today. It looks alright with 1600x1200. However, it's a bit snowy outsite and no sunshine today. We'll obersve it next day when there's sunshine. The CPU load is about 65% and looks better that non-mmal one.

[image]
[image]
[image]
[image]
[image]

piecam, 2015-02-21:
lastfreenick 
 I haven't added "mmalcam_name vc.ril.camera".  Do you see any different without this line?
thanks

lastfreenick, 2015-02-21:
To be honest to you I don't know if or what tgis line does, it was just a line that was specific to motionmmal's example config so I thought it wouldn't garm to add it. Then you add invalid lines to the conf file there usually is an error in the motion.log file but tgere isn't in mine so I guess it does something

lastfreenick, 2015-02-21:
See for yorself, motion example conf (from line 152): https://github.com/dozencrows/motion/blob/mmal-test/configs/motion-mmalcam.conf

piecam, 2015-02-21:
Thanks, I will leave this off for a few days and then add this on to see what differene it makes.

Maybe ccrisan can pitch in and let us know what the meaning for this line, or asking dozencrows.

tks

piecam, 2015-02-21:
on your link,
https://github.com/dozencrows/motion/tree/mmal-test

Known working resolutions:

	352 x 288
	512 x 288
	640 x 480
	768 x 432
	768 x 576
	1024 x 576

Does this affect us? It seems that 1600x1200 is working fine on my recorded pic stills.

piecam, 2015-02-21:
here's a link that touchs a bit on this.
http://www.jvcref.com/files/PI/documentation/html/group___mmal_default_components.html

piecam, 2015-02-21:
define 	MMAL_COMPONENT_DEFAULT_CAMERA   "vc.ril.camera"

piecam, 2015-02-23:
I have a very cold sunshine day today with lights reflecting from snow from the ground. It's still a bit "white bright". However, it's no longer "greeny" and it ajusts well during the day. It has no negative effect on USB cam as well. Both are at 1600x1200.

piecam, 2015-02-23:
[image]
[image]
[image]
[image]

piecam, 2015-02-24:
today, there's no sunshine and the camera looks great 1600x1200. No more "green" and "too white" after using "mmal" version. Will keep it till next week to test that vc.ril.camera.

thanks

piecam, 2015-02-24:
cam1 is pi IR or NoIR cam. Cam2 is usb cam.

piecam, 2015-02-25:
It looks like that my cams are all "too white" and "greeny" again this morning. I am not sure whether it's caused by my turned off my router /switch over night. Pi would have lost network connection to drop files. Not sure whether this is the reason or something else. will try to leave network on for a few days to see.
[image]
[image]
[image]
[image]

lastfreenick, 2015-02-25:
One of my cams is wifi and itcreboits every now and then when connection to the network becomes unreliable but this really should not affect picture quality or cause green tint

piecam, 2015-02-28:
yes, it seems that it's working fine with 1600x1200 with net disconnected or connected. thanks

piecam, 2015-02-28:
not sure about that "greeny" and "white" and why it has happened that time again. so far, in the past few days, we have sunshine,snow and it seems good.

piecam, 2015-02-28:
cpu is always about 60% with "mmal". it looks great.

piecam, 2015-02-28:
with another usb WEb Cam, cpu is about 75%.

piecam, 2015-03-02:
The above is with addition of line "mmalcam_name vc.ril.camera" in thread1.conf. So far it looks great working with "motion-mmal".

Thanks a lot  for both lastfreenick and ccrisan.

piecam, 2015-03-02:
Maybe, it can be integrated into the system by default or can be modified by GUI.

ccrisan, 2015-03-02:
I will add a config entry to allow custom motion options that go into motion.conf and thread-x.conf. That will cover all these mmal needs.

piecam, 2015-03-02:
much appreciated! Great Work!!

lastfreenick, 2015-03-03:
Didn't someone post something about some kernel tweak (in the picam shortfalls thread I think) that would effectively make the cam play nice in higher resolutions? If this would actually work there wouldn't be a need for mmal at all.

piecam, 2015-03-03:
maybe ccrisan has some update on that!
but so far, your "how to" works well.tks

ccrisan, 2015-03-03:
I'm not aware of any kind of kernel tweak that would improve the performance pi camera performance. If anyone knows of such a tweak, please point me to it and I'd be happy to look into it and try to integrate it in motionPie.

lastfreenick, 2015-03-03:
Here it is at the bottom, post by henrik1g https://github.com/ccrisan/motionPie/issues/20

ccrisan, 2015-03-03:
Indeed, those are some module parameters that will keep the video mode instead of switching to stills mode at higher resolutions. The parameters are included in 20150228.

lastfreenick, 2015-03-03:
So if this does indeed work there is no more need for mmal stills mode, I'm still on 215, I will update and check

ccrisan, 2015-03-03:
You'll need to try it on a different SD card. It's still a prerelease so you won't be able to update using the normal procedure.

lastfreenick, 2015-03-03:
Just saw that too, I'll have to wait for the next full release then, my cams are in hard to reach places.

ccrisan, 2015-03-03:
Well let's hope the release won't brick your hard-to-reach PIs then :)

lastfreenick, 2015-03-03:
Every update I do I literally sit there crossing fingers and biting my lip. But until now,  everything has always worked just fine. Very smart decision to put that update functionality into motionpi in the first place because I guess quite a few pis are hanging in some lofty postions ;)

piecam, 2015-03-03:
if you guys can let me know what to change, I can test it out with one pi in house.
Do I have to revert all lastfreenick told me to do?

If you can provide steps, I can test .

tks

ccrisan, 2015-03-03:
@piecam, install [this](https://github.com/ccrisan/motionPie/releases/tag/20150228) version from scratch, on one of your spare SD cards. Then configure the camera to use a high(er) resolution and see if you can feel a difference. Then share your experience with us :)

piecam, 2015-03-03:
will test out and let you guys know soon.

[25 more replies]