ccrisan/motionpie #91: Cisco Linksys WVC-200

piecam, 2015-01-28:
HI,
I have two linksys wvc-200 from a few years back, when it was about $250/pc. I tried on MP, and it doesn't get connected with user/pass.
I tried this link, http://www.lavrsen.dk/foswiki/bin/view/Motion/LinksysWVC200SupportPatch

netcam_url mjpg://netcam-ip:port/img/video.mjpeg 

My adapted version is http://admin:admin@x.x.x.x:80/img/video.mjped.

Any advice?
tks

ccrisan, 2015-03-31:
What http authentication method does it use? Only basic authentication is supported by the motion daemon.

piecam, 2015-04-02:
#EXTM3U
http://admin:zzzzz@192.168.1.21/img/video.asf

This is what I used for VLC to play. However, With firefox, just standard HttP.

thanks

ccrisan, 2015-04-12:
Most likely your camera requires HTTP digest authentication, which is not supported by motion.

closed 2015-04-12 by ccrisan