ccrisan/motionpie #95: Text overlay settings lost when other setting changes are made

barjac, 2015-02-03:
Since there is no "none" option for the text overlays, I save a single space to both settings as "custom text".
This works to remove the overlays, but only until the next reboot or "save" of some other setting (not certain about reboot) when the default text re-appears.
It would be good if there was a NONE option for both labels, as I am using this to stream to ZoneMinder which adds it's own overlay.
Only one pi working as an IP cam so far but building more - thanks for a really useful project!

snagglewest, 2015-02-04:
I'm doing the same with zoneminder and turn off text overlay too. There is a slide button to turn text overlay on and off in at least the last couple of versions.  However, I too have noticed the setting reverting to "on" mysteriously. It's random as far as I can tell I can't reliably reproduce it but I haven't done a whole lot to track it down.  It's a nuisance though.

barjac, 2015-02-04:
Thanks for pointing out the slider switch - I had assumed that was to expand the UI to allow access to the controls rather than to actually enable/disable the feature. I will see how long it stays off now.

Off topic
I wonder if you have found any IR illuminators that work well with pi-noir cameras - I have just tried a Chinese 48 led metal cased unit and am rather disappointed with the very low useful range.

barjac, 2015-02-04:
Re-testing:
Set the Text Overlay slider to OFF and save. Re-boot and the slider is back ON.
Set both text fields to a custom text of one space and save. Re-boot and the default settings return.
Set both text fields to a custom text of one space and save. Change frame rate slightly and Save - this time the Custom text is retained OK.

So it seems to be only the re-boot that resets it.
@snagglewest: IIRC the system re-boots on network faults, so this could explain the randomness.

ccrisan, 2015-02-04:
This seems to be a bug indeed. I will try to fix it in the next release.

ccrisan, 2015-02-08:
The bug has been fixed in motionEye. It will be included in the next motionPie release.

closed 2015-02-08 by ccrisan