ccrisan/motionpie #97: Image won't boot on raspberry 2

floreseken, 2015-02-03:
GREAT piece of software!!

Unfortunately the current image won't boot up on the new raspberry 2.
Will this be easy to fix?

Thanks

ccrisan, 2015-02-04:
I can't tell until I get my hands on a PI2 device, but I can assure you there will be a release compatible with this new device.

trelex, 2015-02-07:
ccrisan, once a rpi2 compatible images has been released do you think we could expect to see MUCH better performance improvements considering the better specification of the rpi2? My understanding is that limitations of the single core arm cpu of the rpi b/b+ is what currently prevents real time streaming over the internet from happening?
thanks

ccrisan, 2015-02-07:
I believe there will be a sensible improvement but I wouldn't go as far as to call it *real time streaming over the internet*.

Roelofloots, 2015-02-09:
Aah man, just got my new raspberry 2 and also cant make it boot...tried building from source, also no luck

ccrisan, 2015-02-09:
Just hold on for a few days until I get my RPI2 board and I'll do my best to release a version for RPI2.

closed 2015-02-15 by ccrisan

floreseken, 2015-03-04:
It works ;-)

Tzaphkiel, 2015-03-12:
Hi, I've tried the last two releases of the RPi2 images and they will not boot for meeither.
 - I tried writing the image using the writeimage.sh from the pi2 folder, 
 - I've tried using dd directly and tried using dcfldd... 

In all cases, I get a :
```
  140+0 records in
  140+0 records out
  ...(147MB) copied...
```
message - which is ok I believe...

When I cfdisk /dev/mm... I see a "W95 FAT16 (LBA)" for the 1st partition... and not "W95 FAT32" or plain "FAT16" ...

When I try to mount the first (vfat?) partition, it is **empty**... 
and hence the booting failing in my opinion (unless I've missed something)...

ccrisan, 2015-03-12:
Try it with another SD card. Also make sure you don't `dd` while the card is mounted. After dd, run the command `sync` so that the kernel buffers are flushed to the disks.

Tzaphkiel, 2015-03-12:
Will try...
Thinking about it and reading the wiki, could it also be the preparation of
the data partition phase on the first boot that takes forever? I'm using a
32gb card...

No but since I don't see anything in the 1st partition when I mount it...
should there be something on it ?

s1th-l0rd, 2015-03-12:
Hi
Have a working boot on Pi 2, finally. I feel around 1.5 sec video stream latency (on 1280x720 30fps with 100% quality)  after the cross compiling with cortex A7 optimizations. Currently streaming through my  mobile WiFi hotspot. I for certain notice a decent bump in performance. :)

@ccrisan How about incorporating WebM support with vp9  --frame-parallel option ??

Tzaphkiel, 2015-03-12:
@s1th-l0rd so you re-compiled and created the partitions yourself ? I'd like to know how you built the partitions to have th Pi2 boot this image.

s1th-l0rd, 2015-03-12:
@Tzaphkiel Sure 
Just to get motionpie running with default Pi2 configuration is quite easy. Download the Code (https://github.com/ccrisan/motionPie/archive/master.zip), navigate to where you have extracted the zip, and on your Linux terminal run 
./build.sh raspberrypi2 

After the image (.img file) is ready ,which takes some time,  use 
sudo dd bs=4M if=image.img of=/dev/mmcblk0(your SD CARD).

If you are using WiFi adapter for pi then use 

sudo gedit /yourmountedsd/etc/wpa_supplicant.conf and add

ssid="YourWifiID"
psk="yourpassword" 

with quotes before parenthesis close
 
Plug the SD back in the Pi, connect WiFi/Ethernet to it. Power on and wait for about a minute, then just punch in the pi's IP in your browser. 
Voila!

 Hope that works, anyways i haven't played around with the partitions yet as they are quite resilient to power-cuts which is why I ditched Raspbian (even though its quite awesome in its own way) in the first place.
Cheers!

ccrisan, 2015-03-12:
There's a prepared image made especially for the Raspberry PI2. You shouldn't compile anything! Check out the [20150228](https://github.com/ccrisan/motionPie/releases/tag/20150228) prerelease and make sure you download the -raspberrypi2- image.

@s1th-l0rd I'm not familiar with "WebM support with vp9 --frame-parallel". Can you walk me through it and if it's something worth integrating in the motionPie image, I'll see what I can do about that.

ccrisan, 2015-03-12:
@Tzaphkiel, yes, the first fat partition must have some files on it. If it doesn't, then the image writing process went wrong.

s1th-l0rd, 2015-03-12:
@ccrisan First of all, thank you so much for this amazing application. About WebM, its relatively a new video format currently promoted by Google and is in application in a lot of new devices and services including the latest Skype for their video streams, of course. 

VP9 is their latest codec offering: 
http://en.wikipedia.org/wiki/VP9  

and, here's what piqued my interest:
http://www.phoronix.com/scan.php?page=news_item&px=VP9-Parallel-Decode

Your application is based on MJPEG, and maybe, just maybe this will yield some more performance+storage improvements.

ccrisan, 2015-03-12:
@s1th-l0rd I know what WebM is :) The specific way you put it in your last comment led me to believe that this is some kind of compile time switch that can improve performance in a way.

Unfortunately there's not much I can do regarding WebM (or any other video formats, for that matter), since this machinery is implemented 100% in the [motion](http://www.lavrsen.dk/foswiki/bin/view/Motion/WebHome) daemon, which currently only supports MJPG. I don't have the necessary experience to work on other video formats and to provide patches to motion.

s1th-l0rd, 2015-03-12:
@ccrisan That's exactly what the article i quoted above led me to believe. Had to confirm :)
Anyways thanks a lot for your time, I still have a hunch WebM can provide some performance improvements. But can't tell unless I make it there. Got my first Pi last month, currently reading a couple of Linux oriented books and fooling around with make, buildroot, and a couple of git repos etc. 
I'm in Love!

Tzaphkiel, 2015-03-14:
@ccrisan the SD card is fine, 'burning' a Volumio image boots it fine.
so not the pie and not he SD-card.
I went throught the process again:
1. download the image from the latest release: 
https://github.com/ccrisan/motionPie/releases/download/20150228/motionpie-raspberrypi2-20150228.img.gz
2. gzip -d motionpie-raspberrypi2-20150228.img.gz
3. file motionpie-raspberrypi2-20150228.img : 
motionpie-raspberrypi2-20150228.img: DOS/MBR boot sector; partition 1 : ID=0xe, active, start-CHS (0x0,32,33), end-CHS (0x2,42,40), startsector 2048, 32768 sectors; partition 2 : ID=0x83, start-CHS (0x2,42,41), end-CHS (0x11,118,37), startsector 34816, 245760 sectors

4. sudo dd bs=1M if=motionpie-raspberrypi2-20150228.img of=/dev/mmcblk0
140+0 records in
140+0 records out
146800640 bytes (147 MB) copied, 10.4596 s, 14.0 MB/s

5. sync
6. take sd card and put in pi2
7. power on
8. on hdmi: rainbow gradient nothing else event after 15minutes...

... If I take the card out, re-mount the p1, I see content ... I'm at a loss !

Oh and @s1th-l0rd I cannot make the master src code, something about my current path being in the LD_LIBRARY_PATH, missing core-dependencies... what are the packages one needs to be able to compile?

ccrisan, 2015-03-14:
@Tzaphkiel can you try this using another sd card (a smaller one), just to rule out any sd card issues.

And on a side note, don't bother compiling it from source, it's a rather complicated process, especially if you're not on a common and up-to-date distro. Let's find out what the root cause of the problem is and I'll make sure to include a fix for this in the next release.

Tzaphkiel, 2015-03-14:
ok but it'll have to wait for next week unless I can find quickly a smaller card.

You think it could perhaps be the expansion routine or the partition types that would not work with so big a card?

this one is a 32GB...

ccrisan, 2015-03-14:
Nope, it cannot be the expansion routine since. Since you're only able to see the rainbow, the kernel never manages to actually boot. A wild guess would be an unsupported SD card, in which case I'd probably need to add some options when compiling the kernel.

Can you try and write the RPI v1 image on your 32GB SD card and boot it on a RPIv1 (if you happen to have one at your disposal)?

Tzaphkiel, 2015-03-14:
... left it at work, not before tuesday :(

will try v1 on v1 later.
will trying v1 on v2 pi make any sense ?

ccrisan, 2015-03-14:
In cmdline.txt: `console=ttyAMA0,115200 kgdboc=ttyAMA0,115200`
In /etc/inittab, uncomment `ttyS0::respawn:/sbin/getty -L ttyS0 115200 vt100`, if you want to be able to log in, in the happy event that it actually works.

Keep me posted.

s1th-l0rd, 2015-03-14:
@Tzaphkiel its only 25 days since I've purged M$ from my household, by choice, and I've been using Vim mostly because I can't figure out how to exit it, so with the compilation part, its either Fedora 21 or I just got lucky ;)
Anyways here's the cheet-sheet from my quest:
1. No need to plug in video.
2. NETWORKING <Ethernet/(pre-configured WiFi)> is must, preferably static IP. Take the sd card and plug it back it again after you dd the image. The card will now be mounted. In the 127MB partition navigate to /etc/wpa_supplicant or whatever you want to config network using sudo gedit. For this GIYBF.    
3. Then access Pi through browser (well i haven't ssh 'ed in as it would be not very helpful)

@ccrisan Don't know why, but my adafruit usb debugging cable is giving me boot loops with

\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\ (fetched through screenlogs)
between every restart.

ccrisan, 2015-03-14:
@s1th-l0rd I'm not familiar with the adafruit usb debugging cable but I doubt it will work with motionPie.

s1th-l0rd, 2015-03-14:
@ccrisan Bump! :confused: So how do you debug your application?
Here I thought [that](http://www.adafruit.com/product/954) cable was more or less the conventional way of debugging a headless Pi.

Tzaphkiel, 2015-03-14:
@s1th-l0rd I for one am using the following ftdi device connected to the serial port on the GPIO header
http://www.electrodragon.com/product/ftdi-basic-wdouble-selections-of-pins/
I have been able to see the RPi console but not interact with it...
For the image, booting & such, the video plugging is to be able to see if the kernel boots up... which it does not... now the serial cable and the debug flags on the cmdline will hopefully tell me why it does not boot.

For the image building, I am using an up to date Arch distro... nevermind, I'll try what @ccrisan told me. I believe that the kernel configuration is not up to the task of booting on such a card... We shall see.

@ccrisan is there a list of the packages used in the image, notably for bonjour, winbind, samba, ftp, etc. I've tried to build an installation with motion eye and was hitting the wall with windows interaction and samba configuration (permissions to access and write to the system).

ccrisan, 2015-03-15:
[this](https://github.com/ccrisan/motionPie/blob/master/configs/raspberrypi_defconfig) could give you an idea of the packages included in the OS.

Tzaphkiel, 2015-03-15:
@ccrisan in the mean time I've made & tried several other distro (volumio, voidlinux, motionpie) images both on a 16GB and 32GB card and the only one that actually boots is the volumio one. All other stay at the 'rainbow screen' and either stall or keep on rebooting. I unfortunately do not have my serial console adaptor here to check the log messages but this in its self is already an indication that there is a problem either with my pi2 or more probably with the boot kernel configuration...

ccrisan, 2015-03-15:
Try downclocking the CPU right after writing the motionPie image. In the config.txt file, change arm_freq=900 to arm_freq=700.

Tzaphkiel, 2015-03-15:
@ccrisan nope that does not work.
I still believe it is a kernel configuration or boot partition type problem
**Volumio rpi2 (booting fine)** 
```
    Disk /dev/mmcblk0: 29.7 GiB, 31914983424 bytes, 62333952 sectors
    Units: sectors of 1 * 512 = 512 bytes
    Sector size (logical/physical): 512 bytes / 512 bytes
    I/O size (minimum/optimal): 512 bytes / 512 bytes
    Disklabel type: dos
    Disk identifier: 0x00043284

    Device         Boot  Start     End Sectors  Size Id Type
    /dev/mmcblk0p1 *      2048  155647  153600   75M  b W95 FAT32
    /dev/mmcblk0p3      155648 3411967 3256320  1.6G 83 Linux
```
**Voidlinux rpi2 (not booting)**
```
    Disk /dev/mmcblk0: 14.9 GiB, 15931539456 bytes, 31116288 sectors
    Units: sectors of 1 * 512 = 512 bytes
    Sector size (logical/physical): 512 bytes / 512 bytes
    I/O size (minimum/optimal): 512 bytes / 512 bytes
    Disklabel type: dos
    Disk identifier: 0x2ba0f00b

    Device         Boot  Start     End Sectors Size Id Type
    /dev/mmcblk0p1 *      4096  124927  120832  59M  c W95 FAT32 (LBA)
    /dev/mmcblk0p2      124928 4192255 4067328   2G 83 Linux
```
**motionPie rpi2 (not booting)**
```
    Disk /dev/mmcblk0: 14.9 GiB, 15931539456 bytes, 31116288 sectors
    Units: sectors of 1 * 512 = 512 bytes
    Sector size (logical/physical): 512 bytes / 512 bytes
    I/O size (minimum/optimal): 512 bytes / 512 bytes
    Disklabel type: dos
    Disk identifier: 0x0c8d3a66

    Device         Boot Start    End Sectors  Size Id Type
    /dev/mmcblk0p1 *     2048  34815   32768   16M  e W95 FAT16 (LBA)
    /dev/mmcblk0p2      34816 280575  245760  120M 83 Linux
```

Maybe this would help finding out why the Volumio image works and not other:
* https://github.com/volumio/build-raspbian-image/tree/master/raspbian
* Armv6 for rpi pi: https://github.com/volumio/RootFS-RaspberryPI
* Armv7: https://github.com/volumio/RootFS

I'm not versed enough in linux system to get something out of it unfortunatly :(

Tzaphkiel, 2015-03-15:
@ccrisan My bad, the version I had purchased was supposed to be a rpi2... the version I have been writing the image for was supposed to be a rpi2 and looks identical but is a B+ V1.2 ...
 
That would explain the not booting for the MotionPie and Void but not the booting for the volumio ...
anyway, thanks for looking into it... 

Rpi b+ v1.2: http://www.rlx.sk/5051-thickbox_default/raspberry-pi-b-8gb-noobs-made-in-uk-model-b-v12-b-plus-4xusb40xgpiousd.jpg
Vs.
Rpi2: http://www.raspberrypi.org/wp-content/uploads/2015/01/Pi2ModB1GB_-comp.jpeg

Apologies again

ccrisan, 2015-03-15:
I'm glad you sorted it out in the end.