ccrisan/motionpie #100: Wifi not working

joostvp, 2015-02-06:
Hi all, 

Perhaps you can help me out! So far the installation is pretty perfect. MotionPie just runs like a sunshine while using an ethernet cable. Unfortunately it isn't working through WiFi.

Before I used MotionPie i also had Raspbian installed and WiFI worked just as expected. So that tells me my dongle is allright.

What I did:
1) Downloaded the package and "burned" it on my SD
2) Connected an ethernet cable to my RPi. The RPi is using the IP: 192.168.2.18
3) Set the correct SSID/PSK
4) Disconnected ethernet and added my WiFi-dongle. On my router I see the RPi is using the IP: 192.168.2.19. Also 182.l168.2.18 remains connected to my router although not connected.
5) When I start the IP 192.168.2.19 my browser keeps loading and eventually doesn't connect.

I checkd the data/etc/wpa_supplicant.conf:
update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
    scan_ssid=1
    psk="noortje1"
    ssid="VGV75197938BD"
}

There is no data to be found in the log file.

Anyone mind helping in solving this issue?

thanks in advance!

Joost

joostvp, 2015-02-06:
As an addition: I also tried creating a static_ip.conf file with:
static_ip="192.168.1.50"
static_gw="192.168.1.1"
static_dns="192.168.1.1"

No result...

snagglewest, 2015-02-07:
It looks like you're missing the key_mgmt=line which tells it what security mode you're using on your router.  I'm using WPA2personal  which is WPA-PSK, you'll need to change that if you're using something else.

My wpa_supplicant.conf is:
------------------------
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    ssid="yourSSID"
    psk="wireless-passowrd"
key_mgmt=WPA-PSK
}
------------------------------------
Note the SSID & psk are in quotes.  The key_mgmt is NOT in quotes.  You're router will assign a new IP for the wireless so that's why you're seeing 2 IP addresses.  Once the router releases the IP it'll disappear, probably 24 hours depending on client lease time you have set in the router.

snagglewest, 2015-02-07:
It looks like you're missing the key_mgmt=line which tells it what security mode you're using on your router. I'm using WPA2personal which is WPA-PSK, you'll need to change that if you're using something else.
My wpa_supplicant.conf is:

update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
ssid="yourSSID"
psk="wireless-passowrd"
key_mgmt=WPA-PSK
}

Note the SSID & psk are in quotes. The key_mgmt is NOT in quotes. You're router will assign a new IP for the wireless so that's why you're seeing 2 IP addresses. Once the router releases the IP it'll disappear, probably 24 hours depending on client lease time you have set in the router.

joostvp, 2015-02-07:
Thx for your reply Snagglewest! 

So I wanted to try it right away.Another problem arises. A problem I didn't have before. When I SSH into the RPi, using Putty, the root password is not the password i have chosen for in the Web-UI.

ccrisan, 2015-02-07:
@joostvp have you rebooted after changing the password? All passwords are updated with the UI value at boot.

joostvp, 2015-02-08:
Ahh... I didn't reboot... how stupid. Now I CAN edit the WPA_SUPPLICANT.CONF file. Unfortunately it isn't possible to edit the /boot/config.txt file. I wanted to turn the led off on my pi. Checking WiFi now...

ccrisan, 2015-02-08:
@joostvp, see [Why do I get Read-only file system when trying to edit a file?](https://github.com/ccrisan/motionPie/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file).

closed 2015-02-08 by ccrisan