ccrisan/motionpie #102: Permission denied when deleting files in output folder via ssh

sonostrano, 2015-02-11:
Hi! Thanks again for this wonderful software.
When I connect to the raspberry and try to use rm -rf * to delete all the folders/file in the folder output, I receive the error "permission denied" or "failure".
Any suggestions?
Thanks

ccrisan, 2015-02-11:
Can I see the exact commands you enter? What credentials do you use to login via SSH?

sonostrano, 2015-02-11:
I logged as admin (I tried also logginf as root) and in /data/output I entered  rm -rf *

ccrisan, 2015-02-11:
What version of motionPie do you run?

sonostrano, 2015-02-11:
20150112

ccrisan, 2015-02-11:
This is odd. What does ls -al /data/output say?

sonostrano, 2015-02-11:
Unable to open -al: no such file or directory
...very strange

I'm using psftp.

by the way... I manually deleted all the files I wanted to delete now

ccrisan, 2015-02-11:
I thought you were using normal SSH (or putty). psftp doesn't allow normal Linux commands.

closed 2015-02-11 by ccrisan

sonostrano, 2015-02-11:
uuu I didn't know, sorry. thank you!