ccrisan/motionpie #103: Issue with motion event being associated with wrong device

ddyeakley, 2015-02-11:
Everything works when I delete all cameras and only add one camera. If I have more than one camera. I have an issue with motion events being associated with wrong device.   I am using motionPie with a couple of Foscam IP cameras (FI8905W).    I receive email  on motion but email says motion is from camera x when in fact if can from camera y (determined by picture and email subject).      Also if I enable feature to show changed pixels and box motion, changed pixels count show up on wrong image.   

I am using 20150208.    I could not test on 20150112 because I would get into a login loop when trying to imput the camera URL.

Any ideas on what might be going on?

ccrisan, 2015-02-12:
The login loop bug is a known bug, fixed in the latest (pre-released) version. In any case, the motion detection mechanism hasn't been touched at all lately so the new version shouldn't make any difference.

What you describe is quite odd. All the motion detection and event triggering is handled by the motion daemon, including the changed pixels text overlay. Can I see a screenshot that exemplifies this issue? Does it happen all the time with no exception?

ddyeakley, 2015-02-13:
I know issues was "odd", I never got 3-4 camera to work with out the problem.     HOWEVER, I deleted the image I was working on and can no longer reproduce.      Consider closed until I can reproduce and provide more details.    I am waiting on a couple Pi 2  and looking forward to testing on the new hardware.

closed 2015-02-15 by ccrisan