ccrisan/motionpie #107: Email not sending photos

unmanagedtn, 2015-02-15:
If i setup an smtp server with the latest release it does not send snapshots with the txt motion detected messages... 

Is there something I am missing? setting?

I am using https://mandrillapp.com to send emails ... I use it for a photobooth and its free for 12k per month emails...

ccrisan, 2015-02-15:
Have you enabled still images? If, at the moment of composing the email, no pictures are found close to the event's moment, no pictures will be attached.

unmanagedtn, 2015-02-15:
yes I did enable still images ... its set to take a shot every 900
seconds... but that should not have anything to do with it right?

http://71.228.245.100/

unmanagedtn, 2015-02-15:
I guess you cant do interval and motion capture frames and the same time I
see...

ccrisan, 2015-02-15:
Nope, you can't.

closed 2015-02-15 by ccrisan