ccrisan/motionpie #108: update crashed

dkelley810, 2015-02-16:
I did an update to 20150215 and a message said it failed. Now it will not boot up at all. I tried to do a power cycle with no luck. Any ideas? I believe that I will have to pull the sd card and redo the image with the new version.

ccrisan, 2015-02-16:
It would be great if I could see the contents of /log directory on the data partition. Updating is a very tricky and difficult mechanism.

dkelley810, 2015-02-16:
that is the partition that contains the output directory, correct. do you 
want what is in the /log/cores and /log/samba?
I emailed them also to you. Some of them are very long. If need be I will post them here. 

log.nmbd - empty
log.smbd - empty
wpa_supplicant.log - empty
wtmp -  
motioneye.logMessages: 

I reached the maximum characters,

ccrisan, 2015-02-17:
@dkelley810, I see a lot of log likes like the following:

    2015-01-25 17:42:01: ERROR: HTTP 405: Method Not Allowed
    2015-01-25 17:42:01: WARNING: 405 HEAD / (74.86.158.106) 34.56ms

or

    2015-01-25 17:45:24: WARNING: 404 GET /muieblackcat (93.158.200.40) 35.06ms
    2015-01-25 17:45:24: ERROR: HTTP 404: Not Found (not found)

or even

    2015-01-25 17:45:24: WARNING: 404 GET //phpMyAdmin/scripts/setup.php (93.158.200.40) 31.12ms
    2015-01-25 17:45:25: ERROR: HTTP 404: Not Found (not found)

They clearly indicate that your public IP address is being scanned and constantly attacked. I assume you have DMZ-ed (or configured port forwarding for) your PI. I believe all this flooding is killing the internal motionPie HTTP server. If the HTTP server is not responding within 120 seconds, the board is automatically rebooted.

Moreover, I am seeing messages like "Network is unreachable" in various logs, messages that indicate a poor or erroneous network link. Without a reasonably good network connection, automatic updating is very unlikely to succeed.

dkelley810, 2015-02-21:
what can i do to fix this? change my IP address thru my router?
Thank You,

ccrisan, 2015-02-21:
I don't know how to help you with this issue. Try different routers, wifi adapters? Use ethernet cable where possible?

dkelley810, 2015-02-21:
Thanks! I am going to redo the sdcard with the latest release. I have not had a lot of time this week, but will get to it today! As usual, thank you for all your help and great work!

closed 2015-02-28 by ccrisan