ccrisan/motionpie #109: Updating problem

sonostrano, 2015-02-16:
Hi! I'm trying to update to last release.
The first time I tried I got an error, the second and third times apparently all went fine. The system says that the update was succeful, but the Current Version is still 20150112.
This is the log

INFO: updating to version 20150215...
2015-02-16 10:25:44: INFO: killing motioneye init script...
kill: usage: kill [-s sigspec | -n signum | -sigspec] pid | jobspec ... or kill -l [sigspec]
2015-02-16 10:25:45: INFO: downloading https://github.com/ccrisan/motionPie/releases/download/20150215/motionPie-20150215.img.gz...
--2015-02-16 10:25:45--  https://github.com/ccrisan/motionPie/releases/download/20150215/motionPie-20150215.img.gz
Resolving github.com... 192.30.252.129
Connecting to github.com|192.30.252.129|:443... connected.
WARNING: cannot verify github.com's certificate, issued by '/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 Extended Validation Server CA':
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... 302 Found
Location: https://s3.amazonaws.com/github-cloud/releases/21212431/bc9e11da-b53b-11e4-8983-aa2c34eaecc2.gz?response-content-disposition=attachment%3B%20filename%3Dmotionpie-20150215.img.gz&response-content-type=application/octet-stream&AWSAccessKeyId=AKIAISTNZFOVBIJMK3TQ&Expires=1424078806&Signature=cDLaOem8LBVkqMhFkO3REzNVlIA%3D [following]
--2015-02-16 10:25:45--  https://s3.amazonaws.com/github-cloud/releases/21212431/bc9e11da-b53b-11e4-8983-aa2c34eaecc2.gz?response-content-disposition=attachment%3B%20filename%3Dmotionpie-20150215.img.gz&response-content-type=application/octet-stream&AWSAccessKeyId=AKIAISTNZFOVBIJMK3TQ&Expires=1424078806&Signature=cDLaOem8LBVkqMhFkO3REzNVlIA%3D
Resolving s3.amazonaws.com... 54.231.244.0
Connecting to s3.amazonaws.com|54.231.244.0|:443... connected.
WARNING: cannot verify s3.amazonaws.com's certificate, issued by '/C=US/O=VeriSign, Inc./OU=VeriSign Trust Network/OU=Terms of use at https://www.verisign.com/rpa (c)10/CN=VeriSign Class 3 Secure Server CA - G3':
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... 200 OK
Length: 44438966 (42M) [application/octet-stream]
Saving to: '/data/.firmware_update/firmware.gz'

     0K .......... .......... .......... .......... ..          0% 1.90K=22s

2015-02-16 10:28:36 (1.90 KB/s) - Read error at byte 43836/44438966 (Connection reset by peer). Retrying.

--2015-02-16 10:28:37--  (try: 2)  https://s3.amazonaws.com/github-cloud/releases/21212431/bc9e11da-b53b-11e4-8983-aa2c34eaecc2.gz?response-content-disposition=attachment%3B%20filename%3Dmotionpie-20150215.img.gz&response-content-type=application/octet-stream&AWSAccessKeyId=AKIAISTNZFOVBIJMK3TQ&Expires=1424078806&Signature=cDLaOem8LBVkqMhFkO3REzNVlIA%3D
Connecting to s3.amazonaws.com|54.231.244.0|:443... connected.
WARNING: cannot verify s3.amazonaws.com's certificate, issued by '/C=US/O=VeriSign, Inc./OU=VeriSign Trust Network/OU=Terms of use at https://www.verisign.com/rpa (c)10/CN=VeriSign Class 3 Secure Server CA - G3':
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... 403 Forbidden
2015-02-16 10:28:38 ERROR 403: Forbidden.

2015-02-16 10:28:38: ERROR: could not download update: Command '['/usr/bin/wget', u'https://github.com/ccrisan/motionPie/releases/download/20150215/motionPie-20150215.img.gz', '--no-check-certificate', '-O', '/data/.firmware_update/firmware.gz']' returned non-zero exit status 8
2015-02-16 10:28:38: ERROR: Command '['/usr/bin/wget', u'https://github.com/ccrisan/motionPie/releases/download/20150215/motionPie-20150215.img.gz', '--no-check-certificate', '-O', '/data/.firmware_update/firmware.gz']' returned non-zero exit status 8
Traceback (most recent call last):
  File "media/data/pi/motionPie/output/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1141, in _when_complete
  File "media/data/pi/motionPie/output/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1162, in _execute_method
  File "/programs/motioneye/src/handlers.py", line 119, in wrapper
    return func(self, *args, **kwargs)
  File "/programs/motioneye/src/handlers.py", line 1287, in post
    result = update.perform_update(version)
  File "/programs/motioneye/src/update.py", line 168, in perform_update
    download(version)
  File "/programs/motioneye/src/update.py", line 104, in download
    subprocess.check_call(['/usr/bin/wget', url, '--no-check-certificate', '-O', _DOWNLOAD_FILE_NAME])
  File "/usr/lib/python2.7/subprocess.py", line 540, in check_call
CalledProcessError: Command '['/usr/bin/wget', u'https://github.com/ccrisan/motionPie/releases/download/20150215/motionPie-20150215.img.gz', '--no-check-certificate', '-O', '/data/.firmware_update/firmware.gz']' returned non-zero exit status 8
2015-02-16 10:28:38: ERROR: 500 POST /update/?version=20150215&_=1424078747114&_username=admin&_signature=bcc02f41f1cb38ba25b5fecc5369be1839fc6408 (151.24.83.155) 173885.11ms
2015-02-16 10:32:25: ERROR: authentication failed for user admin
2015-02-16 10:32:28: WARNING: control hue not found for device /dev/video0
2015-02-16 10:32:30: WARNING: control hue not found for device /dev/video0

ccrisan, 2015-02-16:
It looks as though the github server (the amazon server to be more precise) returned forbidden when trying to download the update. Too overloaded i guess? I'll try to catch and treat this kind of errors in an intelligent way.

sonostrano, 2015-02-16:
You are the best

closed 2015-02-22 by ccrisan