ccrisan/motionpie #110: Administrator Username

snagglewest, 2015-02-16:
I just installed 0150215 with a new image and the "Administrator Username" isn't editable, it's defaulted to "admin".  Everything else is editable including the "Surveillance Username".   Not a huge deal, can it be changed thru SSH?

ccrisan, 2015-02-16:
It can be changed from ssh, but it:s not advisable. It was hardcoded to "admin" to match the unix user admin which is used for ssh, samba, ftp and so on. So don't change it.

snagglewest, 2015-02-17:
Ok, so I have two MotionPie cams, the first I upgraded from the "upgrade" button within MotionPie and it retained my "Administrator Username" as "administrator" which is what I set to previously in v20150112.  On that one the "Administrator Username" is not editable now.  On the second MotionPie cam I upgraded from a new image and on that one the  "Administrator Username" is "admin" which is also not editable now either.  So now I have two MotionPies with different  "Administrator Username".  Not a big deal to adapt to the new format, but how do I change the old one to "admin"?  What path/file do I need to edit to change the "Administrator Username" do get them both "admin" via SSH?  I know I could just re-image the one, but it's mounted and not all that assessable.

ccrisan, 2015-02-17:
I'm sorry for this inconvenience. You can ssh into your PI using admin/your_admin_password and manually edit the motion.conf file:

    nano /data/etc/motion.conf

Change the @admin_username value to admin and reboot. Everything should be fine thereafter.

closed 2015-02-17 by ccrisan

snagglewest, 2015-02-17:
Worked perfectly!

No inconvenience at all, I just needed the correct file to edit.  Didn't want to muck around in there willy nilly with no idea what I was looking for or doing :)

Thanks