ccrisan/motionpie #113: Reboot (hanging MotionPie on Rasp2)

jadijkstra2000, 2015-02-20:
Hi,

I have motion pie running on a rasp2, but after a while it 'hangs'.
Webserver does not respond anymore.
I like to know how I can reboot it through SSH. Can execute the command reboot, but nothing happens.
Init 6 says : Must be run as PID 1, but PID 1 is INIT (root)

Regards,
Johan

ccrisan, 2015-02-20:
`reboot` should do the trick. If it doesn't, then something fishy must have happened there. Can you please attach the zipped contents of /var/log/ directory? Or at least /var/log/messages.

jadijkstra2000, 2015-02-20:
Hi, thanks for you're quick response :)

Well, I tried reboot, as user, admin and root. When you give enter the SSH Session 'hangs' and nothing happens, reconnect to motion pie just opens the console again. So nothing happened. I will try to collect the logs.

jadijkstra2000, 2015-02-20:
A piece of messages:
Feb 20 09:25:07 mp-0a01520d auth.info sshd[16665]: input_userauth_request: invalid user user [preauth]                                    
Feb 20 09:25:09 mp-0a01520d auth.err sshd[16665]: error: Could not get shadow information for NOUSER                                      
Feb 20 09:25:09 mp-0a01520d auth.info sshd[16665]: Failed password for invalid user user from 10.0.0.88 port 64110 ssh2                   
Feb 20 09:25:11 mp-0a01520d auth.info sshd[16665]: Failed password for invalid user user from 10.0.0.88 port 64110 ssh2                   
Feb 20 09:25:12 mp-0a01520d auth.info sshd[16665]: Failed password for invalid user user from 10.0.0.88 port 64110 ssh2                   
Feb 20 09:25:12 mp-0a01520d auth.info sshd[16665]: Connection closed by 10.0.0.88 [preauth]                                               
Feb 20 09:25:14 mp-0a01520d auth.info sshd[16667]: WARNING: /data/etc/moduli does not exist, using fixed modulus                          
Feb 20 09:25:14 mp-0a01520d auth.info sshd[16667]: Invalid user user from 10.0.0.88                                                       
Feb 20 09:25:14 mp-0a01520d auth.info sshd[16667]: input_userauth_request: invalid user user [preauth]                                    
Feb 20 09:25:16 mp-0a01520d auth.err sshd[16667]: error: Could not get shadow information for NOUSER                                      
Feb 20 09:25:16 mp-0a01520d auth.info sshd[16667]: Failed password for invalid user user from 10.0.0.88 port 64112 ssh2                   
Feb 20 09:25:18 mp-0a01520d auth.info sshd[16667]: Failed password for invalid user user from 10.0.0.88 port 64112 ssh2                   
Feb 20 09:26:28 mp-0a01520d auth.info sshd[16667]: Failed password for invalid user user from 10.0.0.88 port 64112 ssh2                   
Feb 20 09:26:28 mp-0a01520d auth.info sshd[16667]: Connection closed by 10.0.0.88 [preauth]                                               
Feb 20 09:26:37 mp-0a01520d auth.info sshd[16670]: WARNING: /data/etc/moduli does not exist, using fixed modulus                          
Feb 20 09:26:39 mp-0a01520d auth.info sshd[16670]: Accepted password for root from 10.0.0.88 port 64185 ssh2                              
Feb 20 09:39:32 mp-0a01520d auth.info sshd[16670]: Received disconnect from 10.0.0.88: 11: disconnected by user                           
Feb 20 09:54:50 mp-0a01520d auth.info sshd[16753]: WARNING: /data/etc/moduli does not exist, using fixed modulus                          
Feb 20 09:54:52 mp-0a01520d auth.info sshd[16753]: Accepted password for root from 10.0.0.88 port 64554 ssh2                              
Feb 20 09:55:25 mp-0a01520d auth.warn login[16764]: invalid password for 'UNKNOWN' on 'pts/8'                                             
Feb 20 10:06:31 mp-0a01520d auth.info sshd[16753]: Received disconnect from 10.0.0.88: 11: disconnected by user                           
Feb 20 10:06:42 mp-0a01520d auth.info sshd[16789]: WARNING: /data/etc/moduli does not exist, using fixed modulus                          
Feb 20 10:06:44 mp-0a01520d auth.info sshd[16789]: Accepted password for root from 10.0.0.88 port 64616 ssh2

ccrisan, 2015-02-20:
Please collect the logs from the sdcard inserted in a (hopefully) Linux PC. I suspect some hardware related issues, such as power supply problems, too much overclocking or a corrupt SD card.

jadijkstra2000, 2015-02-20:
Uhhmm, how can I upload log files? I can only upload PNG, GIF, or JPG :)

ccrisan, 2015-02-20:
Just send them to me over email (ccrisan at gmail dot com)

jadijkstra2000, 2015-02-20:
I have mailed you the log files. I have looked into the files myself, but cannot see strange things...

ccrisan, 2015-02-28:
This is most likely due to SD card errors or other hardware failures.

closed 2015-02-28 by ccrisan