ccrisan/motionpie #114: can't get the resolution higher than 1600x1200 with my pi cam module

cdnr1, 2015-02-20:
i have tryd editing the v412ctl.py but i get 

2015-02-20 18:03:20: WARNING: Read error on 8: [Errno 104] Connection reset by peer
2015-02-20 18:03:20: WARNING: error on read
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/tornado/iostream.py", line 354, in _handle_read
    if self._read_to_buffer() == 0:
  File "/usr/lib/python2.7/dist-packages/tornado/iostream.py", line 421, in _read_to_buffer
    chunk = self._read_from_socket()
  File "/usr/lib/python2.7/dist-packages/tornado/iostream.py", line 402, in _read_from_socket
    chunk = self.socket.recv(self.read_chunk_size)
error: [Errno 104] Connection reset by peer
2015-02-20 18:03:20: WARNING: mjpg client connection for camera 1 on port 8081 is closed
2015-02-20 18:03:20: WARNING: Connect error on fd 17: ECONNREFUSED
2015-02-20 18:03:20: WARNING: Connect error on fd 18: ECONNREFUSED
2015-02-20 18:03:20: WARNING: Connect error on fd 19: ECONNREFUSED
2015-02-20 18:03:20: WARNING: Connect error on fd 20: ECONNREFUSED
2015-02-20 18:03:21: WARNING: Connect error on fd 21: ECONNREFUSED

ccrisan, 2015-02-20:
I hope you're on a RPI2, otherwise such a big resolution doesn't make sense.

The width and height must be multiples of 16 for motion to accept them. Make sure they are.

closed 2015-02-20 by ccrisan

cdnr1, 2015-02-20:
yes i am

cdnr1, 2015-02-20:
crap just checking if you new
solved it