ccrisan/motionpie #116: suggestions, possible coorporation with other distro/developers.

piecam, 2015-02-23:
HI ccrisan,

I have used 4 Pi B+ with your distro and it worked great! 

However, I found out the following other developers/distro.  
http://www.raspberrypi.org/forums/viewtopic.php?f=43&t=63276
https://minibianpi.wordpress.com/

I have loaded it up on minibian and it looks very good with fast frame rate. The GUI may not appeal as yours.

I was wondering whether you guys can reach out to each other and coordinate some development effort together. That would be wonderfult.

Just a thought to share. thanks

ccrisan, 2015-02-23:
It's not gonna happen. The two distros have somethibg fundamentally different at their core: one has motion daemon and the other has the picamera.

closed 2015-02-23 by ccrisan

piecam, 2015-02-23:
HI ccrisan,

sorry, it looks that they have 'motion' as well. I don't know much under hood. Maybe it's worth it for you to take a deep peek. Just some thoughts. Hope you don't mind.tks

ccrisan, 2015-02-23:
They use motion strictly for motion detection and they feed it with jpegs collected every now and then. For the core, they use various RPI camera-specific code and libs.

motionPie on the other hand is built directly on motion and relies on it for 90% of its features. It cannot be replaced without rewriting pretty much everything from scratch.

piecam, 2015-03-01:
HI ccrisan,
sorry to bring this up again, just want check with you.
Your motionPie works as it's intented for both picam and usb web cam!
However, it looks to me that this RPI Web Interface project, specifically with raspMJPEG , they can handle much higher rate of frames and with higher resolution of captures (video/images). Maybe I missunderstood it.

Anyway, it would be great that you accept some donation from me and you spend some time and effort in poking around their system. Probably that you can create a fork from it and get it working with your "motioneye" incorporating their function. If you can make it work with other usb web cam, that would be fantastic; otherwise, have this as a seperate project sole for Pi CAM, for the sake of high frame rate and high resolution.

It seems that they use raspMJPEG interact with cam and motion monitors raspMJPEG's local stream, then issue command to raspMJPEG to record stills or video.

My understanding might be wrong. hope you don't mind. thanks again.

ccrisan, 2015-03-01:
At the core of motionPie OS there's a python project called motionEye. At the core of motionEye project there's a program called motion, which does all the video-related work. While I control motionPie and motionEye, I do not (nor I am willing to) control the motion daemon.

All setups that perform better on a Raspberry PI, they do it simply because they take advantage of the PI camera (API and whatnot). This is not the case with motion (not even with motion-mmal).

piecam, 2015-03-05:
HI ccrisan,

Hope you don't mind that I am bringing up this topic again for the sake of exploring technology in this area.

I have used 2 more pis testing out this RPI Web Interface with Robert's fork. Then, suddenly, with more understanding of how it works, I have the following idea.

I posted on Robert's web forum.  I'd like to see how you think about it. Maybe it's doable. thanks in advance.

Is this possible?

Another point is that we could leverage different threadx.conf under motion. Then, thread1.conf deals with Pi Cam specifically; and thread2.conf deals with an attached USB cam, same for thread3.conf with another usb cam.
If this works, then, we can achieve similar result as MotionPie, function wise. That's how motionpie is using utilizing different threads for different cam.
So, raspMJPEG is triggered by motione's thread1.conf and motion-mmal (or motion itself) is triggered by thread2.conf for usb cam.

You think this is doable?
Many many thanks.

https://github.com/roberttidey/RaspiCam/issues/1#issuecomment-77363192

ccrisan, 2015-03-05:
I don't have the time to work on such a task. If anyone else is willing to take a shot at it, I'd be happy to review the work and try to integrate it in the mainline motionPie.