ccrisan/motionpie #117: Camera Issue, can be seen but won't add.

chrislongden, 2015-02-25:
Hi,

I've been using motionPie and it's awesome! However I wanted to get a webcam that included IR to do night vision and I found this:

http://www.amazon.co.uk/gp/product/B00LAYY4J8?psc=1&redirect=true&ref_=oh_aui_detailpage_o01_s00

This camera can work on it's own or you can press a button to make it work as a webcam. Now I've tested the camera on my laptop and it works. I connected it to the pi and it shows as "USB Video Class" when I'm adding a camera but once I click add I just get the loading "dots" and it just sits there.

Now I know this camera works with the pi as I've seen a few reviews of others connecting it for there own CCTV setup.

Any thoughts or debugging ideas that I can do?

chrislongden, 2015-02-25:
For reference I have the camera and a USB Wifi adapter plugged directly into the pi... do you think it may need a USB powered hub?

ccrisan, 2015-02-25:
It might be related to a bug in the driver/v4l2-ctl command that makes the command `v4l2-ctl --list-formats-ext` to hang. A similar problem has been reported a couple of times. I'll try to workaround this issue and kill the command after a certain timeout.

chrislongden, 2015-02-25:
Is there a way that I could implement this fix for the time being before you put it live in a next update?

ccrisan, 2015-02-25:
I myself am not really sure how to implement that right now. I'm sorry but you'll have to wait for a next release.

ccrisan, 2015-02-25:
As a temporary workaround, if you're familiar with Linux, you could remove v4l2-ctl binary and replace it with a script that simply echoes the supported resolutions, one per line, in a format like 640x480.

chrislongden, 2015-02-25:
Yeah the camera itself only supports up to and including 640x480, which is the res I need. Familiar-ish with linux, guess I'll just have to play around and see (unless someone else reads this and can point me in the direction of the steps to do this).

ccrisan, 2015-02-28:
It should be fixed (well, worked around) in 20150228.

closed 2015-02-28 by ccrisan