ccrisan/motionpie #120: pairing thread-1 to camera 1

piecam, 2015-02-28:
HI,
As I have Pi Cam and USB cam for some Pis. I have also have Pi cam with 2 additional USB cams.

Normaly, Thread1 goes with Pi Cam, thread2 with a usb cam and thread3 goes with another usb cam.
When I do reboot, some times,  the Camera 1 goes with wrong threads.

Is there a way that we can lock up the pairing (configuration threads with specific cam)? As I was tuning up the setting, for example, in Recommended Settings topic, I have put a few different lines for each threads, such as resolution, pi cam specific parameter, etc.

thanks

ccrisan, 2015-02-28:
It depends on the order in which the kernel detects the video devices. I believe it could be forced in a way or another by udev rules. I'll have to look into it. Any help is appreciated.

piecam, 2015-03-03:
If you need me to capture some message, dmesage, etc. or test something, I can help with that. :-)
thanks. for looking into this.

ccrisan, 2015-03-04:
I actually need someone with udev skills to help me configure a persistence rule for the video devices.

ccrisan, 2015-04-12:
This problem has been solved by using /dev/v4l/by-id/* paths to access devices rather than the generic /dev/video* ones.

closed 2015-04-12 by ccrisan

jepolch, 2015-08-24:
Can you please explain - or give a link - how to change the device names? I tried to change them (rearrange the order) in the UI, but it will only let me copy the link, not write. I have three cameras and would like to rearrange them:

/dev/v4l/by-id/usb-046d_0825_9AB222B0-video-index0
/dev/v4l/by-id/usb-046d_0824_BB6CE8A0-video-index0
/dev/video0

Thank you.

ccrisan, 2015-08-24:
Your only option is to remove them and add them in the correct order. There are plans to implement UI support for reordering and organizing your camera frames, but. as I always say, don't hold your breath.

jepolch, 2015-08-24:
OK, I'll give that a try. It will be kind of tricky since the Raspi cam is #2. Thanks for the help.