ccrisan/motionpie #121: script for mounting a remote CIFS/SMB

piecam, 2015-02-28:
HI ccrisan,
I have 4 Pis with your MP running great with the following remote SMB/CIFS mounting setup.
# @network_share_name Recording
# @network_server 192.168.100.10
# @network_username cam
# @storage_device network-share
# @network_password xxxx

target_dir /data/media/motioneye_192_168_100_10_recording_cam
picture_filename %Y-%m-%d/Frontdoor_Out/%H-%M-%S

//192.168.100.10/Recording
                        870.9G    346.5G    524.4G  40% /data/media/motioneye_192_168_100_10_recording_cam

I am also trying another PI with Minibian with Rpi Web Interface (the one runs off raspMJPEG). I was wondering whether you can shed some quick tips in doing a remote share on that. That would survive at boot when there's no network up yet and when there's a reboot or network loss.

I would appreciate it very much. I know that it's a bit of off topic. But, you're the expert and I  thought that you don't mind sharing .

thanks

piecam, 2015-02-28:
In that Rpi Web Interface, which uses raspiMJPEG, they have this in memory,
preview_path /dev/shm/mjpeg/cam.jpg
image_path /var/www/media/image_%04d_%04d%02d%02d_%02d%02d%02d.jpg
video_path /var/www/media/video_%04d_%04d%02d%02d_%02d%02d%02d.mp4
status_file /var/www/status_mjpeg.txt
control_file /var/www/FIFO

I would like to link this image_path /var/www/media to network share. something like,
//192.168.100.10/Recording
870.9G 346.5G 524.4G 40% /var/www/media/motioneye_192_168_100_10_recording_cam
thanks in advance.

piecam, 2015-02-28:
found this, http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=57113

But, if you could help with a simple guides it would be much appreciated! tks

Yours in MP works perfectly. As yours work in a debian with motion+motioneye, I thought that you would have similiar set up that I can use. much appreciated!

ccrisan, 2015-02-28:
I'm sorry but I did not understand what you are trying to accomplish. Do you have problems with a network share going down due to network/connectivity issues? Do you want to set up a single NAS using Linux and make your motionPies output their media files on that NAS?

piecam, 2015-03-01:
HI ccrisan,
No, your motionpie works flawlessly so far. 
I have another spare Pi with pi cam and I have minibian installed together with RPI Web INterface.
I'd like to mount a SMB at boot.
this is what I have now and it seems works.
root@raspberrypi:~# cat /etc/fstab
/dev/mmcblk0p1 /boot vfat defaults 0 2
/dev/mmcblk0p2 / ext4 errors=remount-ro,noatime 0 1
//192.168.100.10/Recording /var/www/media cifs username=yyy,password=xxxx,workgroup=workgroup,users,auto,user_xattr 0 0

My concern are,
1. By your experience, will I bump into a situation that mounting happens before network is ready and thus cause mounting failure? Will /etc/fstab try repeatedly forever or some time or at all?
2. If there's network discoonect, will mounting be resumed after network is connected again?
3. What is your way (script) of setting this smb share properly.

thanks in advance.

ccrisan, 2015-03-01:
I now understand your question.

1. Normally you won't. The Debian scripts that take care of the init dependencies will make sure the network is up before mounting net shares.
2. Nope, it doesn't work that way.
3. I take care of this issue by manually managing the mount points with a [python module](https://bitbucket.org/ccrisan/motioneye/src/b50d24175d74005b0f11c2415faa3185b5967900/src/smbctl.py?at=master) and checking up on them every now and then.

closed 2015-03-01 by ccrisan

piecam, 2015-03-01:
much appreciated!