ccrisan/motionpie #126: UVCVideo: Failed to query

chrislongden, 2015-03-03:
I get UVCVideo: Failed to query once I've added my camera - obviously it's "unable to open video device"...

I know the cam I have works with r-pi's but I'm guessing maybe it doesn't with motionPie??

ccrisan, 2015-03-03:
1) what version of motionPie are you currently using?
2) what kernel did you test it on (when it worked)?
3) what is the driver it uses? (can you do a lsmod in an OS where it works?)

chrislongden, 2015-03-03:
I'm using the current pre-release 20150228 and the camera has worked on a previous install of raspbian

lsusb: 
Bus 001 Device 009: ID 04d6:205c Mentor Graphics 

lsmod: 
pi@raspbmc:/$ lsmod
Module                  Size  Used by
bcm2708_wdog            2340  1 
ipv6                  251548  37 
xt_state                 664  1 
iptable_mangle           760  0 
iptable_nat             1736  0 
nf_conntrack_ipv4       8860  2 
nf_defrag_ipv4           760  1 nf_conntrack_ipv4
nf_nat_ipv4             2358  1 iptable_nat
nf_nat                  8820  2 nf_nat_ipv4,iptable_nat
nf_conntrack           54280  5 nf_nat,xt_state,nf_nat_ipv4,iptable_nat,nf_conntrack_ipv4
iptable_filter           624  1 
ip_tables              10592  3 iptable_filter,iptable_mangle,iptable_nat
x_tables               10532  4 ip_tables,xt_state,iptable_filter,iptable_mangle
snd_soc_tas5713         4128  0 
snd_soc_pcm512x         6592  0 
snd_soc_wm8804          6064  0 
8192cu                458968  0 
joydev                  8264  0 
uvcvideo               64860  0 
videobuf2_vmalloc       1896  1 uvcvideo
videobuf2_memops        1116  1 videobuf2_vmalloc
videobuf2_core         25072  1 uvcvideo
videodev               95848  2 uvcvideo,videobuf2_core
media                   8656  2 uvcvideo,videodev
snd_soc_bcm2708_i2s     4668  0 
regmap_mmio             1796  1 snd_soc_bcm2708_i2s
snd_soc_core           99000  4 snd_soc_pcm512x,snd_soc_wm8804,snd_soc_tas5713,snd_soc_bcm2708_i2s
snd_compress            6124  1 snd_soc_core
regmap_i2c               824  4 snd_soc_pcm512x,snd_soc_wm8804,snd_soc_core,snd_soc_tas5713
snd_pcm_dmaengine       2428  1 snd_soc_core
regmap_spi               968  3 snd_soc_pcm512x,snd_soc_wm8804,snd_soc_core
i2c_bcm2708             3596  0 
snd_pcm                62044  2 snd_soc_core,snd_pcm_dmaengine
snd_page_alloc          2984  1 snd_pcm
snd_timer              17252  1 snd_pcm
snd                    41760  4 snd_soc_core,snd_timer,snd_pcm,snd_compress

ccrisan, 2015-03-04:
What is the model of your webcam? I cannot seem to find it online by the vendor and product ids.

chrislongden, 2015-03-04:
This is the webcam: http://www.amazon.co.uk/gp/product/B00LAYY4J8?psc=1&redirect=true&ref_=oh_aui_detailpage_o01_s00

ccrisan, 2015-03-31:
Can you try the latest version and see if it works with your camera?

closed 2015-05-16 by ccrisan