ccrisan/motionpie #128: Network Share

thegnnu, 2015-03-05:
Hi First thanks for the usb to network driver I asked about as I can now install a image from a model A great.
I have been having a problem getting Network Share setup even this morning made a new image from 12/1/2015 and updated it.
I first tried to connect to a NAS Dlink box without success so I have used a model B to make a fileserver and install smb
and I can access all files using either 192.168.0.22/motionpie/data or win8 picks up on map a drive
\\FILESERVER\motionserver\data or \nas\video these have been placed into the 3 correct locations no user or password are required but I get nothing and of course the camera turns off.
I have even tried a user name and password but still the same.
Below is the error lines from the logs all timed after 12.00am today
first the composite display gives this error message
CIFS VFS:cifs_mount failed w/return code = -113
CIFS VFS: Error connecting to socket aborting operation
parts of logs  I tried 4 time 2 without password and 2 with user and password and on 2 different servers setups
logs
messages log
Mar  5 12:00:08 mp-7cd363ff auth.info sshd[379]: WARNING: /data/etc/moduli does not exist, using fixed modulus
Mar  5 12:00:09 mp-7cd363ff auth.info sshd[333]: WARNING: /data/etc/moduli does not exist, using fixed modulus
Mar  5 12:05:28 mp-7cd363ff auth.info sshd[13413]: WARNING: /data/etc/moduli does not exist, using fixed modulus
Mar  5 12:05:32 mp-7cd363ff auth.info sshd[13413]: Accepted password for root from 192.168.0.38 port 50708 ssh2
Mar  5 12:05:33 mp-7cd363ff auth.info sshd[13424]: WARNING: /data/etc/moduli does not exist, using fixed modulus
Mar  5 12:05:33 mp-7cd363ff auth.info sshd[13424]: Accepted password for root from 192.168.0.38 port 50712 ssh2

motioneye.log
2015-03-05 11:58:28: ERROR: failed to mount smb share "//192.168.0.22/motionserver" at "/data/media/motioneye_192_168_0_22_motionserver_root"
2015-03-05 12:01:22: WARNING: control hue not found for device /dev/video0
2015-03-05 12:01:22: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_0_22_motionserver_root/data'
2015-03-05 12:01:22: WARNING: control hue not found for device /dev/video0
2015-03-05 12:01:22: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_0_22_motionserver_root/data'
2015-03-05 12:01:51: WARNING: control hue not found for device /dev/video0
2015-03-05 12:01:51: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_0_22_motionserver_root/data'
Unable to find suitable address.
2015-03-05 12:01:57: ERROR: failed to mount smb share "//192.168.0.22/motionserver" at "/data/media/motioneye_192_168_0_22_motionserver"
2015-03-05 12:01:57: WARNING: control hue not found for device /dev/video0
2015-03-05 12:01:57: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_0_22_motionserver/data'
2015-03-05 12:02:12: WARNING: control hue not found for device /dev/video0
2015-03-05 12:02:12: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_0_22_motionserver/data'
Unable to find suitable address.
2015-03-05 12:02:18: ERROR: failed to mount smb share "//192.168.0.22/motionserver" at "/data/media/motioneye_192_168_0_22_motionserver_root"
2015-03-05 12:02:18: WARNING: control hue not found for device /dev/video0
2015-03-05 12:02:18: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_0_22_motionserver_root/data'
2015-03-05 12:02:51: WARNING: control hue not found for device /dev/video0
2015-03-05 12:02:51: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_0_22_motionserver_root/data'
Unable to find suitable address.
2015-03-05 12:02:58: ERROR: failed to mount smb share "//192.168.0.22/nas" at "/data/media/motioneye_192_168_0_22_nas"
2015-03-05 12:02:58: WARNING: control hue not found for device /dev/video0
2015-03-05 12:02:58: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_0_22_nas/video'
2015-03-05 12:03:09: WARNING: control hue not found for device /dev/video0
2015-03-05 12:03:09: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_0_22_nas/video'
Unable to find suitable address.
2015-03-05 12:03:15: ERROR: failed to mount smb share "//192.168.0.22/nas" at "/data/media/motioneye_192_168_0_22_nas_root"
2015-03-05 12:03:15: WARNING: control hue not found for device /dev/video0
2015-03-05 12:03:15: ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_192_168_0_22_nas_root/video'
Unable to find suitable address.
2015-03-05 12:03:34: ERROR: failed to mount smb share "//192.168.0.22/nas" at "/data/media/motioneye_192_168_0_22_nas_root"

Thanks
TerryR
.

ccrisan, 2015-03-06:
I'm not really sure what you are trying to do. Are you talking about the SMB server on your motionPie? Are you talking about a NAS? A raspberrypi/raspbian-based SMB server?

If you have trouble configuring a network share as a storage for your motionPie, check [this](https://github.com/ccrisan/motionPie/wiki/Troubleshooting#network-shares) article.

thegnnu, 2015-03-08:
Calin thanks for the reply I was confused abt the reference to smb server etc but have now downloaded the image 28/2/15 and all explained and working well
Thanks for a great bit of software
TerryR

closed 2015-03-08 by thegnnu