ccrisan/motionpie #129: ssh connection refused

kratostav, 2015-03-05:
Every time when i try to connect via ssh i get this error:
[image]
network is over WLAN (EDIMAX EW-7811UN Wireless USB Adapter stick)
webUI works fine
IP is static:
[image]
Model RaspberryPi B+
What should i do?

ccrisan, 2015-03-05:
How about the services section? Can I see a screenshot of that?

kratostav, 2015-03-06:
[image]

ccrisan, 2015-03-08:
How did you write the image to the sdcard? Can you send me th

kratostav, 2015-03-08:
with win32diskimager
how will i get this without ssh?

ccrisan, 2015-03-08:
Insert your sdcard into your pc and, while running Linux, acces the third partition.

You'll find the log directory in the root of that partition. Just compress it and send it to me over by email.

You cannot do that using windows.

closed 2015-03-31 by ccrisan

airtonGit, 2015-10-01:
Hi here too, can't connect, stuck in version 20150317...

ccrisan, 2015-10-01:
Write the image again from scratch and do not disconnect it at first boot within the first 5 minutes.

airtonGit, 2015-10-01:
Hi, I cant make ssh connection to work, but ok I just take the sd card 
and updated manually, now everything is just fine, Camera Fast Mode is 
so cool! I just cloned the repo and I hope be smart enough to contribute 
some way. thanks!

bluevalien, 2015-10-24:
I just made my fist blog account to acknowledge this solution.
"Write the image again from scratch and do not disconnect it at first boot within the first 5 minutes."
I did this and now ssh is working without an issue. I have no idea why it was not working before but I think this needs to be an important note for anyone who uses this image.  
Note: You have to use the Administrator username (root did not work for me)
unrelated: This works great! I had been trying to develop my own and gave up now that I've found this. Kudos!  Let me know if there is any way to contribute to the cause. (note: I'm Canadian so my money is practically worthless)