ccrisan/motionpie #133: Is There any way to use Java VM with your motionPie?

idealist33, 2015-03-08:
Hi I sent my question to you via Gmail but you've not answered...so i write something here..

i want to send messages to my android phone when motion is detected. 
To send messages I want Java Vertual Machine in motionPie
However motionPie does not have java vm...

i tried to build my own iso image but it is very hard for me to do that

i downloaded build source from the URL https://github.com/ccrisan/motionPie/wiki/Building-From-Source
and then After commanding 'build.sh [raspberrypi]' shows 'bad substitution'.

I tried to command 'make' but i have only rootfs.tar file  in output/image folder 
I have only a little knowledge about rootfs..it is super difficult for me.

Do you have any plan to add Java VM to motionPie image for the next update ???

ccrisan, 2015-03-08:
I really hope you didn't actually type the square brackets around "raspberrypi". The command should read: `./build.sh raspberrypi`.

idealist33, 2015-03-08:
sorry... it doesn't work at all..here is my ssh command window...

ub@ub-VirtualBox:~/motionFin/motionPie$ sh build.sh raspberrypi
build.sh: 11: build.sh: Bad substitution

=====================================================
is there any way to use java without rebuild your source???

ccrisan, 2015-03-08:
use `bash` instead of `sh`, or simply type `./build.sh ...`

ccrisan, 2015-03-08:
I'm sorry, for now please just use `bash` in front of it. If you have `dash` or any other shell set as default on your system, the `./build.sh` won't work. I'll make sure to fix this.

idealist33, 2015-03-08:
Oh bash works!
if i want to use java, should i do 'make menuconfig' and add jamvm and 'bash ./build.sh raspberrypi' ??

ccrisan, 2015-03-08:
Yes, that should normally do it, although I did not test it.

idealist33, 2015-03-08:
Thanks a lot!!! I'll try~ :)

closed 2015-03-31 by ccrisan