ccrisan/motionpie #134: error with open wifi network, no passkey

rleyden559, 2015-03-08:
I have a Pi A so I followed the procedure to add wpa_supplicant.conf and static_ip.conf to /data/etc under Ubuntu.
The Pi successfully does the first boot and I can configure further in the web interface.
However, under, network there is a red exclamation point in "wireless network key".  My entry for the AP has "key_mgmt=NONE"  in wpa_supplicant.conf.  Blank is not accepted in the web interface.

The web interface will not save any of the changes as long as the red exclamation point is there.
I tried entering None (no quotes) and the Pi rebooted but I couldn't find it on the network.  I ended up re-flashing the SD card and now back to the same point.

Rich

ccrisan, 2015-03-09:
There's no way to have this working, for now. I will add support for psk-less networks in the next version.

rleyden559, 2015-03-09:
Thanks.  I found that if I don't expand the network tab I can at least make
and save changes to other fields other fields.  However, the static IP can
only be set or changed by ssh or writing to the SD card.

I'm OK for now.  

 

<https://github.com/ccrisan/motionPie/issues/134#issuecomment-77815798> .
<https://github.com/notifications/beacon/AK2mILzBrLGg4Gipa9avkaILEe2pWt4Cks5
nzVGqgaJpZM4DrXsq.gif> 

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

ccrisan, 2015-03-14:
The next release will no longer require a wifi network password (key)

closed 2015-03-14 by ccrisan

rleyden559, 2015-03-14:
Great!

I really appreciate your approach to MotionPie.  Compared to similar RasPI
projects it is much closer to an embedded device that "just works" .  I've
used Icamview webcam server for the last 5 years.  It's a closed proprietary
system but clearly is based on a relatively low powered linux system.  While
it lacks many features one may wish, each release is indefinitely stable "as
is".  One aspect of the Raspberry Pi I've grown tired off is that
applications tend to die completely if even a minor update is attempted.  

 

Rich

 

<https://github.com/ccrisan/motionPie/issues/134#issuecomment-80548787> .
<https://github.com/notifications/beacon/AK2mIGMlQovKFt2jJ_WcDwR_tlV58dwhks5
n1FEMgaJpZM4DrXsq.gif> 

---
http://www.avast.com