ccrisan/motionpie #135: Randomly stops streaming

barjac, 2015-03-09:
I have currently two pi model b's one wireless and one wired running with a zoneminder server on the network. (and I'm assembling 3 more). All options for motion detection etc. are disabled in both with only video streaming 'ON' for zoneminder to use. The cameras are pi-cams
This works fine for maybe a day or so, then the stream from one or the other will vanish leaving a blank blue image in ZM. (This has happened many times for the wireless camera but only once for the wired, however it's not wireless range related as the wireless one is only a few inches from the access point at present under test)
If I access the motionpie web interface the initial screen shows a frozen old image, however if I click the image it expands and starts working real time, and zoneminder re-acquires the stream.
I see nothing relevant in the motionpie logs when accessed over ssh.
Is there anything that you can suggest or add that would detect this situation and re-start the stream?

ccrisan, 2015-03-09:
You say that if you enter the web UI, the image will refresh when you click on the frame. Does it also refresh automatically after a few seconds? It's common for motionEye to serve an older (cached) image when you load the page.

snagglewest, 2015-03-09:
This is the same issue I'm having with zoneminder with the same setup, only streaming on. I haven't noticed the stream stopping on the Pi, but then again I haven't been looking for it but I've found a reboot on the Pi fixes it until the next time.   it's only every week or so on wireless.  
https://github.com/ccrisan/motionPie/issues/98

barjac, 2015-03-09:
@ ccrisan
It certainly does not refresh itself after a few seconds - I think I looked at it for maybe 20 secs  but I will wait longer when it next happens before clicking the image frame. The frozen frame was in the dark some time in the night, I discovered that it had frozen around lunch time today, so once frozen it needs human intervention :\
@ snagglewest
Glad it's not just me ;)

barjac, 2015-03-09:
2015-03-09 11:22:19.388140	zmdc	4294	ERR	'zmc -m 14' exited abnormally, exit status 255	zmdc.pl	
2015-03-09 11:22:19.344634	zmc_m14	13146	ERR	Failed to pre-capture monitor 14 (0/1)	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zmc.cpp	260
2015-03-09 11:22:19.333552	zmc_m14	13146	ERR	Unable to connect to camera	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zm_remote_camera_http.cpp	1082
2015-03-09 11:22:19.328767	zmc_m14	13146	ERR	Unable to connect to the remote camera, aborting	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zm_remote_camera_http.cpp	115
2015-03-09 11:22:19.221540	zmdc	4294	ERR	'zmc -m 14' exited abnormally, exit status 255	zmdc.pl	
2015-03-09 11:22:19.177953	zmc_m14	11491	ERR	Failed to pre-capture monitor 14 (0/1)	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zmc.cpp	260
2015-03-09 11:22:19.166856	zmc_m14	11491	ERR	Unable to connect to camera	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zm_remote_camera_http.cpp	1082
2015-03-09 11:22:19.157000	zmc_m14	11491	ERR	Unable to connect to the remote camera, aborting	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zm_remote_camera_http.cpp	115
2015-03-09 11:22:19.147834	zmc_m14	11491	ERR	Connection dropped by remote end	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zm_remote_camera_http.cpp	614
2015-03-09 11:22:18.335399	zmc_m14	11491	ERR	Connection dropped by remote end	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zm_remote_camera_http.cpp	614
2015-03-09 11:22:15.826142	zmc_m14	11491	ERR	Connection dropped by remote end	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zm_remote_camera_http.cpp	614
2015-03-09 11:22:13.314916	zmc_m14	11491	ERR	Connection dropped by remote end	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zm_remote_camera_http.cpp	614
2015-03-09 11:22:10.803660	zmc_m14	11491	ERR	Connection dropped by remote end	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zm_remote_camera_http.cpp	614
2015-03-09 11:22:08.292594	zmc_m14	11491	ERR	Connection dropped by remote end	/home/baz/BLD_Mga5_6/zoneminder/BUILD/ZoneMinder-1.28.0/src/zm_remote_camera_http.cpp	614

ccrisan, 2015-03-09:
@barjac can I see the motion.log and motioneye.log files? I would also need to know the approximate moment whem this happened.

barjac, 2015-03-09:
I have looked at those on both cameras and nothing jumps out as being relevant.
I did notice a message implying that syslog was being used for logs, but there is no /var/log/syslog. 
It's a while since I used a system not using systemd ;)
Can we increase logging verbosity somehow?

ccrisan, 2015-03-09:
I'm talking about /var/log/motion.log and /var/log/motioneye.log. The rest is irrelevant to this issue.

barjac, 2015-03-09:
In the wireless camera motionpie.log I found this which looks promising but it's going back a while:

2015-02-17 12:00:07: WARNING: control hue not found for device /dev/video0
2015-02-17 12:00:13: WARNING: control hue not found for device /dev/video0
2015-02-17 12:00:45: WARNING: control hue not found for device /dev/video0
2015-02-17 12:00:50: ERROR: Stream is closed
Traceback (most recent call last):
  File "media/data/pi/motionPie/output/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1115, in _stack_context_handle_exception
  File "media/data/pi/motionPie/output/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 302, in wrapped
  File "/programs/motioneye/src/mjpgclient.py", line 95, in _on_connect
    self.write(b"GET / HTTP/1.0\r\n\r\n")
  File "media/data/pi/motionPie/output/target/usr/lib/python2.7/site-packages/tornado/iostream.py", line 205, in write
  File "media/data/pi/motionPie/output/target/usr/lib/python2.7/site-packages/tornado/iostream.py", line 580, in _check_closed
StreamClosedError: Stream is closed
2015-02-17 12:00:50: WARNING: control hue not found for device /dev/video0
2015-02-17 12:01:34: WARNING: control hue not found for device /dev/video0
2015-02-17 12:01:40: WARNING: control hue not found for device /dev/video0
-----------------snip------------------
then lots more of the control hue warnings and a few
 "ERROR: authentication failed for user null"
 right up to today.

The motion.log has only:

[root@mp-7ef789d7 ~]# cat /var/log/motion.log
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] motion_startup: Motion trunkREV561 Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (EMG)

I will wait until it happens again and try to get more useful complete logs

ccrisan, 2015-03-09:
Well, "Stream is closed" indicates that the TCP connection to the mjpeg server (motion) is broken. motionEye normally tries to reestablish that connection each time it finds it broken. I'd like to see what's happening during the problem. It would be great if you could set up remote SSH access for me to your motionPie next time you find it hanging like this.

barjac, 2015-03-09:
Yes I should be able to do that - just wait - it won't happen for a week now ;)

barjac, 2015-03-13:
Preparing for this I'm trying to change the root password back to the original which I had set to a personal one.
When logged in as root I try with:
[root@mp-7ef789d7 ~]# passwd
Changing password for root
New password: 
Bad password: similar to hostname
Retype password: 
passwd: can't create '/data/etc/shadow+': File exists
passwd: can't update password file /etc/shadow

Logging in as admin:
[admin@mp-7ef789d7 ~]$ passwd
Changing password for admin
Old password: 
New password: 
Bad password: similar to hostname
passwd: password for admin is unchanged

In both the above I use my personal password to log in - how do I change it back to the default hex?
I'm confused about the root, admin and user accounts - is 'admin' as settable in the webUI the root password? I have changed it there and still it will only accept my personal PW at ssh login using:
ssh root@192.168.1.86
Any ideas?

ccrisan, 2015-03-13:
The admin unix account is an alias for root (has the same UID). The password for root/admin is set in the UI, using the admin password field. Even if you manage to change it manually using passwd, it will be overwritten at reboot using the one set in the UI.

barjac, 2015-03-13:
If I set a new admin PW in the UI, save and reboot, I can log in with it to the UI.
Attempts to log in over ssh as either root or admin with it fail and the old PW still works.
So still getting nowhere :\

barjac, 2015-03-13:
OOps now I know what Close and Comment means :\ Not the browser!

ccrisan, 2015-03-13:
@barjac I assume you use the last 20150228 version. Can you take a look a the boot sequence (on a monitor) and see what it says when it tries to set the root/admin password. There's a small message indicating this precise step, but unfortunately the boot sequence is not logged to a file.

barjac, 2015-03-13:
It was the previous version but no change after updating.
I will try to look at the boot, but will have to rig up a composite monitor - I have no HDMI displays.

barjac, 2015-03-14:
I got to see the boot messages albeit at very low res, but just readable:
########
-------------snip------------
Setting up data partition
Setting custom root/admin password
Starting wpa_supplicant: done
------------snip------------
Setting smb admin password: done
-----------snip-------------
########
So no errors, but I notice there is no ": done" on the root/admin as there is later on the smb pw setting.
I looked down the whole output and could see nothing that looked wrong.

ccrisan, 2015-03-14:
There should be no "done" after that message. Everything looks clear to me. So what you are saying is that you're still able to login using SSH/putty with your old password, while the new password is only valid for the UI?

What about other services, such as smb of ftp. What do they accept? the old or the new credentials?

barjac, 2015-03-14:
Yes seems so - what I set in the UI works in the UI but only the old PW works in ssh.
I'm not used to playing with ftp (even less smb), but this did not authenticate with the UI password:
--------------------------
[baz@jackodesktop ~]$ wget --ftp-user=root --ftp-password=7ef789d7 -v ftp://192.168.1.86
--2015-03-14 15:02:45--  ftp://192.168.1.86/
           => ‘.listing’
Connecting to 192.168.1.86:21... connected.
Logging in as root ... 
Login incorrect.
----------------------------------
[baz@jackodesktop ~]$ smbclient -L 192.168.1.86 -U root
Enter root's password:      (entering nothing or any PW produces the same output)
Domain=[MOTIONPIE] OS=[Unix] Server=[Samba 3.6.24]

        Sharename       Type      Comment
        ---------       ----      -------
        sdcard          Disk      SD Card Output Directory
-------------------------------------------------------------------------
Maybe you could suggest some better test commands ;)

ccrisan, 2015-03-14:
Do you have any special characters in your password (other than alphanumeric)? If yes, what are those characters. They might not get passed well to the passwd command.

barjac, 2015-03-14:
No, the current one set in UI is 7ef789d7 and works in UI. The one that currently works in ssh is 5 lowercase letters followed by 2 digits - no spaces, punctuation or anything odd.

ccrisan, 2015-03-14:
7ef789d7 looks a lot like a PI serial number. Did you manually set this password in the UI?

barjac, 2015-03-14:
Yes - trying to revert to the default setting, however if I change to anything else in the UI it only works in the UI not ssh.

ccrisan, 2015-03-14:
I'm really sorry I cannot reproduce this problem. Would you by any chance be available for a remote debugging session? I would really like to see what's going on there. We can work either using teamviewer or your can set up port forwarding so that I can SSH into your PI and see what's going on.

If you're ok with this, just contact me by mail.

barjac, 2015-03-14:
I'm just setting up another SD card in it to try to duplicate the issue, if I can I will mail you with details, but even if I can't it may be useful to let you have a dig around in the original one - I will keep it for testing.

barjac, 2015-03-14:
OK here's what I just did:
1. Create new SD card (using motionPie-20150102.img as I could not get into the latest version at all other than ping)
2. Check I could access UI without PW and ssh using pi serial PW - all OK
3. Created totally new admin password, saved and re-booted
4. Checked it worked in UI and ssh - yes it did! (this is where I will stop in future :\ )
5. Changed the PW in the UI back to the pi serial, saved and rebooted
6. Checked access over ssh with pi serial PW all OK
7. Tried to access UI as admin and no password (including none) now works but I can view without password in UI as I didn't set a normal user/password.
I will re-test (after re-writing the SD) but avoid using the pi-serial as PW this time.

ccrisan, 2015-03-14:
The serial number as password in the UI has no special significance - it should behave just like any other password. Also, there's no point in reverting back to the initial password (empty for the UI and S/N for SSH) and that's why you cannot do that.

Regarding step 7, after explicitly setting the admin password to the S/N, I really doubt the UI will let you log in as administrator with no password. Are you sure about number 7?

barjac, 2015-03-14:
Yes 7 was badly worded in a hurry, it was meant to mean:
7. Tried to access UI as admin and now none of the passwords (including none)  work.
I can view the UI without password as I didn't set a normal user/password, but can not access any settings.

Thanks for reminding me that of course the default UI PWs are blank. I was not trying to get back to that situation.
I will try the latest version and repeat the 1-7 steps again - if it locks me out of the admin UI I will open a new bug as this is now way off topic.

barjac, 2015-03-14:
Using motionpie-raspberrypi-20150228.img on a new Pi and SD I have tested changing the admin PW over and over including using the pi-serial hex without problems, so this problem seems fixed in this version.
One caveat - this has no camera attached yet.

Thanks for your patience and help.

I will use this new Pi/SD to continue testing the real bug report here, and will email with IP etc. if it stops streaming.

barjac, 2015-03-27:
Since updating to the latest version at the time of the previous post, I have not seen this issue and the cameras have bee up 24/7, so closing this issue with fingers crossed.
Thanks again for a great project!

closed 2015-03-27 by barjac